OpenTracksApp/OpenTracks #200: Manually starting GPS and opening a recorded track

dennisguse, 2020-04-28:
**To Reproduce**
1. No recording is running
2. TrackListActivity: go to menu and start click start GPS
3. Open an already recorded track
4. TrackDetailActivity opens, but shows UI for recording a track
(should just open and just present the recorded track in non-recording mode).

Bug was introduced while adding the "resume track"-feature.

**Technical information**
 - Version v3.6.0 and upwards

dennisguse, 2020-04-29:
As of v3.7.1, the start gps feature is deactivated as a workaround.

rgmf, 2020-04-29:
It slipped away for me. I'm going to take a look at it.

dennisguse, 2020-04-29:
One alternative is actually to remove this functionality completely.
I don't use it that often as my GPS sensor is in general rather fast.

Any input on this?

rgmf, 2020-04-29:
> One alternative is actually to remove this functionality completely.
> I don't use it that often as my GPS sensor is in general rather fast.
> 
> Any input on this?

I've never used it because I thought it was there for turn Android Location on (a shortcut to access the Location Android menu) and for inform to the users that Location was on or off.

The problem is that TrackDetail detect that user want to Resume when: service is running, trackId! = -1and is not recording another track. I thought the only path to start service was record button. I'm sorry :'(

To fix it I would need to differentiate between:
1.- Click on Start GPS: only start service.
2.- Click on record button: start service and a recording.

Start GPS option starts service to speed up the GPS sensor availability. Is it needed? Many years ago GPS take much time but nowdays I think this isn't a problem. I would remove this functionality.

dennisguse, 2020-04-29:
@pstorch @mondlicht-und-sterne @Pilou8 @Flavien06 @edern-appere @slackline @lennartschoch @eliotb @sporksmith @PhiPhi84120 @pieplu @ml- @pes4murai  @serpico @vanitasvitae @marcocamma @rgmf 
What is your opinion about this feature?
For simplicity: it would be great if just upvote / downvote this comment :)

sporksmith, 2020-04-29:
> What is your opinion about this feature?

I've never used it, and don't see a need for it.

> For simplicity: it would be great if just upvote / downvote this comment :)

Done - downvoted :)

ml-, 2020-04-29:
1. I honestly don't even know if my GPS sensor is good, bad, fast or slow. How would one measure it?
2. If the GPS sensor is started with the recording, does the time start when pressing the button or when GPS is ready?

(haven't been running for a while due to injury, have to catch up with the updates)

slackline, 2020-04-29:
I've up voted it as a feature because getting a GPS fix isn't simply a function of how good the hardware in the phone is, its contingent on getting several fixes from the GPS satellites ([a minimum of four](https://gis.stackexchange.com/questions/12866/why-does-gps-positioning-require-four-satellites)) in order to position you.  Depending on where you are this can have a little lag (e.g. buildings, valleys) so being able to start GPS and get a fix before starting activity is I feel a useful feature.

@ml- : A useful app for checking the GPS on your phone is [GPSTest](https://play.google.com/store/apps/details?id=com.android.gpstest&hl=en_US)

eliotb, 2020-04-29:
I wasn't aware of the Start GPS option.  If I want to get an early fix, I use SatStat app which also allows me to request AGPS data update.

PhiPhi84120, 2020-04-30:
downvote

I'm using an old smartphone for that, so I need to start GPS 5mn before using opentracks.

commit 363bb76e (2020-04-30): Manually starting GPS and opening a recorded track bug fixed. Fixes #200.
commit 16017414 (2020-04-30): Manually starting GPS and opening a recorded track bug fixed. Fixes #200.
commit 2815aa4b (2020-05-01): Merge pull request #203 from OpenTracksApp/starting_gps#200

Manually starting GPS and opening a recorded track bug fixed. Fixes #200.
closed 2020-05-01 by rgmf