OpenTracksApp/OpenTracks #202: Feedback for speed and cadence sensors

dennisguse, 2020-04-29:
We need some testing for the speed and cadence sensors.
Development was done using a Wahoo CADENCE (although not attached to a bicycle).

_What we need:_
1. What kind of sensor did you test?
2. Is a sensor selectable in the settings?
3. Does a sensor report data?
4. Does the data look correct?

DevRonin, 2020-04-30:
I ordered Garmin sensors 2 speed & cadence.
Hope to test them soon.

sisyphosloughs, 2020-05-02:
I can report positive Tests for

- Hart Rate: Wahoo Tickr (Model: WFBTHR02)
- Speed: Wahoo Speed (Model: WFRPMSPD)
- Cadence: Wahoo Cadence (Model: WFPODCAD2)

Tested:
- OpenTracks 3.7.1-Playstore
- Faiphone 3: Android 9, Build 8901.2.A.0118.2020406

Pairing with Bluetooth settings in Android was not necessary. I simply added the sensors within the setup of the OpenTracks app.

dennisguse, 2020-05-10:
I just found and fixed a bug in the computation (time difference between sensor data was rounded to full seconds). Fix will be in v3.7.3
Affects cadence as well as speed.

dennisguse, 2020-05-13:
Just released v3.7.3 containing the fix. 
Works for me.

dr-m, 2020-06-04:
Hi! New old user here. Years ago, I tried to use MyTracks on my previous phone (SonyEricsson Xperia Active) and the ANT+Sport based sensor Garmin GSC-10. I am happy that all the cloud integration has been removed, and I think that Bluetooth (traditional or LE) is the future.

Yesterday I made a quick experiment on v3.7.4-d1f3791f- (version code 3396) from F-Droid. I used a Meilan SC sensor that works nicely with AAT (Android Activity Tracker). There was no obvious way to tell if the speed sensor worked (AAT auto-detects the wheel size and displays "Speed GPS" or "Speed S"). The cadence sensor was displaying incredible values (sometimes in excess of 200 rpm, instead of the correct value that should have stayed somewhere between 80 and 100 rpm).

I seem to remember that already MyTracks had trouble with the cadence measurements. It would use some weird averaging, resulting in the displayed cadence dropping slowly until the averaging time window runs out. In AAT, the display would drop to 0 rpm within one or two seconds of stopping pedaling. If the interval between the previous 2 cadence pulses was 1 second (corresponding to 60 rpm), then I think that it should suffice to wait 2 seconds before displaying 0 rpm.

I also tested a Bluetooth LE based heart rate sensor for about 30 seconds. It displayed credible bpm.

It might be nice to have a ‘database’ of multiple speed and cadence sensors. Ideally, OpenTracks would know the sensors on all my bicycles, and it would automatically associate the trip with the correct bicycle and sports type (say, touring, mountain biking, tandem biking). The same could be extended to other sensors (say, step counter on running or climbing shoes).

dennisguse, 2020-06-04:
@dr-m cadence is transmitted after one complete rotation only, but we actually can reduce the "no data received" timeout from 5s to 1 or 2s.
And the database idea is great. Will add it tomorrow for the equipment I tested with.

serpico, 2020-06-13:
Tested with 3.7.4 from F-Droid on a Samsung Galaxy Core Prime SM-G361F Android 5.1.1
   
-  What brand/model did you test?
**Decathlon|Geonaute|Kalenji Dual HRM Belt ( ZT26D molded in the plastic on the back of the sensor case, on the right of the battery cover, once you separate the sensor case from the belt, unclipping the snap fastener )**
-  Is a sensor selectable in the settings?
**Yes**
-  Does a sensor report data?
**Yes**
-  Does the data look correct?
**Yes**

Note : Should work with the cardiosport TP3 since it seems to use the same ZT26D but I didn't test it.

dennisguse, 2020-06-13:
@dr-m Can you provide a link to the sensor you have?
I did not find it.

closed 2020-06-14 by dennisguse

ebellumat, 2020-09-05:
Hello! I know this issue is closed, but...

 I need some help with a bluetooth device.

https://www.amazon.com/MEILAN-Wireless-Bluetooth-Computer-Waterproof/dp/B07QCPLVFC

I connect to him but I can't parse the retrieved data from this little device.

These device returns UInt8Array data.

Anyone can help me to retrieve the cadence data from this device?

I want to get the RPM but i don't understand the returned values to convert.

Regards, Erasmo.

[image]

[image]

commit 0a685a66 (2020-06-14): Added list of tested Bluetooth LE sensors.
Fixes #202.