OpenTracksApp/OpenTracks #205: JSON export feature

eiabea, 2020-04-30:
**Is your feature request related to a problem? Please describe.**
I am currently working on a self hosted analysis dashboard based on nestjs, influxdb and grafana and use the data from opentracks' export feature to extract the plain numeric information from the kmz files description part with some regex functions. This approach works, but is pretty error-prone.

**Describe the solution you'd like**
My solution would be an additional option in the export feature to export the plain numeric data (avg. speed, distance,...) in a normalized format (preferential JSON)

**Describe alternatives you've considered**
Can not think of any

dennisguse, 2020-04-30:
GeoJSON should work.
I just need to check if it support sensors...

eiabea, 2020-04-30:
I am not familiar with GeoJSON, but as far as i understand this format is used for saving polygons and lines on a map, but can it also store metadata like overall average speed or total distance of the whole trip.
Sorry if my feature request is unclear, but i wouldn't need the geo data, just the metadata of the trip, but booth would be fine too, maybe something like this:
```json
{
  "meta": {
    "averageSpeed": 23,
    "distance": 1337,
  },
  "geojson": {
    "type": "Feature",
    "geometry": {
      "type": "Point",
      "coordinates": [125.6, 10.1]
    },
    "properties": {
      "name": "Dinagat Islands"
    }
  }
}
```

eiabea, 2020-04-30:
The units for each parameter would be nice to follow the [SI base unit](https://en.wikipedia.org/wiki/SI_base_unit)  :nerd_face: 
Speed: `m/s`
Time: `s`
Length: `m`

dennisguse, 2020-04-30:
You just want the statistics? Not the geodata?

eiabea, 2020-04-30:
@dennisguse exactly, just everything which is currently in a human-readable form in the `Description` field of the `KML` file

But i really like the idea of having a combined JSON export with statistics and geojson if this is possible

dennisguse, 2020-10-31:
I guess this is out of scope for now.
We are still struggling with the internal data structures and I guess we will focus on GPX/KML.

closed 2020-10-31 by dennisguse