OpenTracksApp/OpenTracks #210: How can I add Wahoo speed and cadence sensors?

sisyphosloughs, 2020-05-02:
Hi, I really like the description of the App. Simple and privat. Exactly that what I'm looking for! But I don't know how I can use my sensors. Maybe a few words in the Help section could make it easier. Maybe it is not a bug, but some explanation would be nice, how users can add sensors. THX!

**Describe the bug**
I have not found a way to add my Wahoo sensors to OpenTracks.

**To Reproduce**
1. Pair Wahoo Sensors.
2. Turn the wheehl & turn the crank: Sensors are blinking blue, so they are on
3. Go to Settings of OpenTracks.
4. In the 'Bluetooth' section only hart frequency is dipslayed.
5. In the 'Help' I did not found any entry for activating the 

**Technical information**
 - Device: Fair Phone 3
 - OS: Android 9
 - Version: 8901.2

dennisguse, 2020-05-02:
These two sensors were added in v3.7.0. However, it is not yet on F-Droid.
But it should happen in the next days. If you don't want to wait you have to compile it yourself or get it from the PlayStore.

But we should add some text to the help section.
Thanks for the reminder.

sisyphosloughs, 2020-05-02:
Thank you for that! I have just tried the play store version, and I was able to connect one sensor. Just one question: Is it necessary to pair the devices? At the moment I'm using the devices with Locus, which doesn't need a pairing. If pairing is necessary a short hint for other users could help.
Regards

dennisguse, 2020-05-02:
@sisyphosloughs Pairing is not necessary (at least not that I am aware of).
Can you report your results (which sensors and were they working) here: #202?

That would be great.

closed 2020-05-03 by dennisguse