OpenTracksApp/OpenTracks #215: Deleting tracks takes unexpected long

eiabea, 2020-05-04:
**Describe the bug**
Deleting a single track takes about 10 seconds.

**To Reproduce**
1. Open up OpenTracks
2. Long-press on a single track to mark it
3. Click on the "3 dots"-menu on the top right corner
4. Click on delete
5. Wait...

**Technical information**
 - Device: Nexus 5X
 - OS: Stock Android 8.1.0
 - Version: v3.6.2

dennisguse, 2020-05-04:
@eiabea Interesting.
In this case, we need some more infos.

* How big is the track that you are deleting?
  Can you export it to KMZ and report the size in MB?
  Also how long is the track?
* Does the track contain pictures?
* Actually, does this also occur for very short tracks?
* What is the size of OpenTracks that is reported in the Android app settings (the one where you can "Force close")?
* When did you install OpenTracks (approx. which version)?
  There might have been an issue with the database.

PS: It is definitively not "now you can go and get a coffee"-feature - at least it was not intended :)

eiabea, 2020-05-04:
> How big is the track that you are deleting?
Can you export it to KMZ and report the size in MB?
Also how long is the track?

The track is very small (sometimes when i open up the app it starts recording right away, which creates a new track i stop immediately), to be exact: 824 Bytes

> Does the track contain pictures?
No

> Actually, does this also occur for very short tracks?
Yes

> What is the size of OpenTracks that is reported in the Android app settings (the one where you can "Force close")?

App size: 9.08 MB
User data: 14.43 MB
Cache: 106 kB
Total: 23.61 MB

> When did you install OpenTracks (approx. which version)?
There might have been an issue with the database.

I started using the app end of 2019, maybe December

eiabea, 2020-05-04:
> PS: It is definitively not "now you can go and get a coffee"-feature - at least it was not intended :)

Sorry, i am not quite sure what you mean, but i am ready to help nailing this issue down! :muscle: 
Should i record the logcat output while deleting?

dennisguse, 2020-05-07:
Thanks for the infos.
Can you try to delete the app data (or re-install OpenTracks)?
Just to understand if there something wrong with the database.

Just export und import your data again.

If you can compile OpenTracks, you can also try to add logcat statements yourself - but at the moment there are none (to the best of my knowledge).

dennisguse, 2020-05-25:
Please re-open if still valid.

closed 2020-05-25 by dennisguse

eiabea, 2020-05-25:
Oh sorry, i switched to another Nexus 5x and there it works like expected :+1: