OpenTracksApp/OpenTracks #228: HR sensor disconnect

DevRonin, 2020-05-11:
I have a Polar H10 HRM, dual bluetooth.
Is connecting simultaneously to both: my phone (OpenTracks) and Bryton 530.
But while is still connected to Bryton, I found it disconnected from my phone and no way to connect it back.
AAT has a menu where you can scan for sensors and choose which one to connect.
How this can be done in OpenTracks?
Pairing sensors from the Bluetooth menu is not working.

dennisguse, 2020-05-11:
OpenTracks should just reconnect while recording (it opens a GATT request when started and Android should take care to reconnect). What you can try is to go to the settings and select the Polar again as heartrate sensor.
And sensors don't need be paired, just set in OpenTrack settings.

dennisguse, 2020-05-13:
@DevRonin Any new insights?

DevRonin, 2020-05-18:
Yep, I had a new workout yesterday.
Started OpenTracks, started bluetooth , no connection.
Options -> selected HRM (was already selected) but still no connection.
Name of HRM did appear on main view, but no reading.
It never show any reading until the end of workout (5h).

Same time, I started my Bryton 530, sensor detected at startup right away.
HRM is a Polar H10, dual Bluetooth, it can connect to two devices simultaneously.

And yes, another time was working fine with both  (Bryton and OpenTrack).

Right now I did another test, HRM was working just fine.
Closed app, closed BT, open again, not working.
Selected HRM from settings menu (alredy selected) - not working.
Interesting, after pairing from BT menu, it was working.

If is any log I can send  ...( I know how to work with adb, device is rooted).

closed 2020-05-18 by DevRonin

dennisguse, 2020-05-18:
@DevRonin Then I take a LogCat (by mail to dennis.guse@alumni.tu-berlin.de).

It sounds as if we have a problem decoding the H10 transmitted data (as you say the device name was shown and this only happening after data was received).

If you would like take a look yourself, the decoding is happening in ` BluetoothUtils.parseHeartRate();`.

DevRonin, 2020-05-20:
I have a new hint:
If sensor is not connected, searching for devices to pair in bluetooth menu seem to find sensor inside the application and data (pulse) is shown. No need of pairing.
But seem strange to need this (some times) in order to connect the sensor.