OpenTracksApp/OpenTracks #233: Can't connect LIVALL Bluetooth Cadence sensor.

programmin1, 2020-05-21:
**Describe the bug**
I cannot connect a LIVALL Cadence sensor.

**To Reproduce**
1. Go to Settings
2. Click on Cadence
3. Only "None" available.

**Technical information**
 - Device: LGMP260
 - OS: 7.0
 - Version latest on FDroid.

programmin1, 2020-05-21:
What's worse, the instructions here give "Connection refused" https://elinux.org/RPi_Bluetooth_LE

dennisguse, 2020-05-21:
Could it be that this sensor does not support the Bluetooth Profile "Cadence and Speed"?
At least, I saw several complaints about not working with Strava.

So, the actual question from my side is: what does this sensor send?
Can you check with BLE Explorer to figure out what it can do?

dennisguse, 2020-05-25:
@programmin1 any more insights?

closed 2020-06-09 by dennisguse