OpenTracksApp/OpenTracks #235: Graphs improvement

ddceca, 2020-05-22:
First of all, thank you very much for this software.

I have a couple of suggestions to improve the elevation and speed graphs. I'm not an experienced coder, but I believe they shouldn't be hard to implement:

****

1) Better adjustment of the vertical limits of the graph.

Even if you run at an almost constant speed of, for example, 11 km/h (never below 10 or above 12 km/h), the graph goes from 0 to 25 km/h. A similar thing happens with the elevation. In my opinion, this renders the graphs almost useless, since all you can see is a slightly rippled curve about the middle of the graph, but you can't really say where you ran faster/slower and, specially, how fast you ran.

To overcome that, vertical zooming would be great, but I understand it may not be easy to implement. As an easier alternative, I suggest:

- Top limit: make it equal or slightly higher than the max values (both in Elevation and speed/pace).

- Bottom limit: ideally, it should be equal or slightly lower than the min values. However, I've noticed that all my speed curves begin at zero, even if I start recording while I'm already running. Side suggestion: could the very first data point be excluded from the graphs?

****

2) Could we have a horizontal line drawn representing the average speed/pace? This way, you can easily see where you ran faster/slower than your average. Perhaps you could draw the average speed in the "by distance" graph, and the average moving speed in the "by time" graph.

****

3) Please, increase the number font size. The top titles are fine, but the numbers in the axes are too little in my opinion.

I hope these suggestions are easy to implement. Thank you very much in advance.

ddceca, 2020-06-04:
So, what's your opinion on these suggestions? Are they doable, interesting, etc?
Thanks a lot!

dennisguse, 2020-06-04:
@ddceca: really cool ideas.
All of them doable (only average speed might be slightly more complicated).

My issue is at the moment relatively simple: lack of time.
So, we (probably) do it some when in the future...

If you have time, please contribute

ddceca, 2020-06-06:
To be honest I don't have much spare time either, but my main problem is that I have little coding experience in general, and zero experience in this environment. If I cloned the code and tried adding something, I don't even know how to get an installable APK from there for testing purposes.

Anyway, thanks a lot for this app!

dennisguse, 2020-06-06:
@ddceca Just use AndroidStudio.
First video I found on youtube: https://www.youtube.com/watch?v=emKtYldlVjw

dennisguse, 2021-04-13:
Regarding:
> Better adjustment of the vertical limits of the graph.

The code can handle this already.
We just have to change how we compute the values and not use zero as default.
Also we may need to make the pre-configured intervals relative to the starting point and not zero.

closed 2021-05-26 by dennisguse