OpenTracksApp/OpenTracks #239: Bluetooth or BLE?

PrzemoF, 2020-05-26:
**Describe the bug**
I cant get any of my BLE devices to show up in OpenTracks. Speed/cademce combo sensor connects to phone, but doesn't show as available. Heart rate sensor doesn't allow to connect to the phone. Both devices use BLE standard - I wrote deivers for them as part of OpenCyclingComputer project. I can provide documentation is that helps.

**To Reproduce**
1. Go to Bluetooth section
2. Try to connect to BLE device
3. Device not on list

**Technical information**
 - Device: Galaxy S10 and 2 BLE sensors
 - OS: stock android
 - Version 9.0

dennisguse, 2020-05-27:
What devices are you talking about?
What protocol do they use?

OpenTracks only supports:
* GATT: Heart Rate Service
* GATT: Cycling Speed and Cadence Service

dennisguse, 2020-05-30:
@PrzemoF We need more infos here.

closed 2020-06-09 by dennisguse

PrzemoF, 2020-06-09:
I missed the previous notification - sorry for that.
The sensors are from TACX, they should work as they are consistent with the BLE/GATT standard. What kind of info do you need? That's for the S&C sensor: https://firszt.eu/uncategorized/lezyna-speed-cadence-sensor-lezyne-sc-249-and-btle-py/
Some more info about the hardware here: https://firszt.eu/uncategorized/bluetooth-ble-gatttool-and-almost-all-those-numbers-explained/

My code from another project here:
https://gitlab.com/PrzemoF/Open-Cycling-Computer/-/blob/v0.14/code/src/plugins/ble_hr.py#L149
https://gitlab.com/PrzemoF/Open-Cycling-Computer/-/blob/v0.14/code/src/plugins/ble_sc.py#L160

I'll try again, maybe it's "user problem" :-)

dennisguse, 2020-06-09:
At least it looks like as if it should work :)

dennisguse, 2020-06-13:
@PrzemoF And did it work?

PrzemoF, 2020-06-20:
Unfortunately no. There is nothing showing up on the available devices list. 
[image]
[image]
[image]
The S&C sensor  device definitely work (that's from my laptop):
`$ sudo hcitool -i hci0 lescan
LE Scan ...
FD:DF:0E:4E:76:CF Lezyne S&C 249
FD:DF:0E:4E:76:CF (unknown)
FD:DF:0E:4E:76:CF Lezyne S&C 249
FD:DF:0E:4E:76:CF Lezyne S&C 249
`
Potentially there is a problem with my phone. I'll try to find some app that allows to test BLE on android.

dennisguse, 2020-06-20:
BLExplorer works well: https://f-droid.org/app/org.ligi.blexplorer

dennisguse, 2020-06-20:
@PrzemoF can you also check the service that your sensor announces? OpenTracks while scanning. The filter is defined here: https://github.com/OpenTracksApp/OpenTracks/blob/924ef3313ebcedf2100343ad5a50467ef3ba704d/src/main/java/de/dennisguse/opentracks/services/sensors/BluetoothConnectionManager.java#L160

dennisguse, 2020-10-06:
@PrzemoF Did you ever figure this one out?