OpenTracksApp/OpenTracks #240: 2 Feature Requests: Countdown, Stop button lock

developersx3, 2020-05-27:
1) Before the tracking starts, it would be great to have a possible countdown.  This could be further enhanced by allow the countdown timer length to be chosen in settings.

2) Accidental clicking of stop. I've seen this on other running apps that require you to press an "unlock" button before stopping a recording to prevent accidental stopping of a track.  An alternative method might be requiring a long press to stop the recording.

I've switched over to OpenTracks recently and its really great.  Low footprint and very functional for my runs.

dennisguse, 2020-05-27:
Valid requests! Can you split it into two issues?
Easier to work with for us...

dennisguse, 2020-05-30:
@developersx3 Why do you want a countdown (What is your use case)?
To wait for first GPS fix?

developersx3, 2020-05-31:
On my phone which is inside my armband before a run, registering a touch is sometimes difficult.  After pressing start thru the armband, you may need to readjust the position of the band.

Screens with LCDs have bad visuals at angles. Additionally looking at a dark mode screen (black background with a red button) is harder, especially when looking at it thru a foggy armband cover, against a sun.  Countdown gives you an audio queue that you have successfully pressed the start button.

When stating a run, I prefer to build a pace when starting for closer to accurate results.

Additionally this might now only give you a better GPS fix as well as proper bearing calculation