OpenTracksApp/OpenTracks #241: Feature Request: Stop button lock

developersx3, 2020-05-27:
Accidental clicking of stop. I've seen this on other running apps that require you to press an "unlock" button before stopping a recording to prevent accidental stopping of a track.  An alternative method might be requiring a long press to stop the recording.

dennisguse, 2020-05-30:
@developersx3 Any suggestions how this should look like?
And what is your preferred solution?

developersx3, 2020-05-31:
Not really a UX guy but as I've seen it with my old running app, the start button is converted to a pause and a larger LOCK buttons appears between the pause and stop button.  Both buttons are disabled until the LOCK button is pressed to unlock them.  If nothing is pressed for a couple of seconds, it relocks.

rgmf, 2020-05-31:
> Not really a UX guy but as I've seen it with my old running app, the start button is converted to a pause and a larger LOCK buttons appears between the pause and stop button. Both buttons are [...]

This approach sounds good. A setting option to protect accedintally pressing buttons and a third button in the middle to unblock the TrackController.

dennisguse, 2020-05-31:
What about press and swipe from center either to the left (pause) or right (stop)?
A 3rd button (at least in my head) would not be beautiful, or?

developersx3, 2020-06-01:
That sounds like a good solution.  More fluid.

ronajon, 2020-08-07:
this would be a good feature. 
i have my phone in a softcase on my steering handle (mountainbiking), and too often some accidental stop (from the protector on the screen? maybe some water drops?) accidentally stops recording. 

I sometimes end up with 5 parts of my ridden track, and need to combine them afterwards

dennisguse, 2020-08-07:
We really need to do something here - it also bothers me quite often...

Two things:
1. Do really need the pause functionality?
  I never used it so far and nowadays, we can also resume a track.
2. What about like the alarm clock does on alert or the incoming call: press center and swipe either left to right.

BUT I have not clue to build such a thing :)

@ronajon You can also resume a recorded track (thx to @rgmf).
Take a lock at the context menu.

pstorch, 2020-08-07:
I use the pause button sometimes. Stop/Resume would have the same effect, but then I would prefer to have the Resume as a big button at the bottom and not hidden in the menu.

dennisguse, 2020-08-07:
@pstorch Then let's stick with the pause button.

We know the following options so far:
1. add a third button to unlock the other pause/stop buttons
2. add something that needs to be swiped (alarm / incoming call handling)
3. add a confirmation dialog
4. remove pause/stop button when shown on the lock screen (if possible)
5. Move the buttons to another tab (would require large UI refactoring)

Any other suggestions?

rgmf, 2020-08-07:
The third button to unlock could be a swip button to unlock (actived buttons = true) and lock (actived buttons = false). What I don't know is where to put this third button in the UI.

pstorch, 2020-08-08:
I saw in another app a short animation on the buttons. You had to hold the button for a while (e.g. 2 seconds) and during this time a an outline circle surrounded the button. When the circle got closed, the button action got  executed.
So similar to swipe, but without needing more space on the UI.

pstorch, 2020-08-08:
I've made a proposal. If you hold the stop button an animation starts and at the end the stop is executed. If you release the stop button earlier, then the stop is canceled.

In this example I've pushed the button two times too short and the third time long enough to stop the recording:
[image]

pstorch, 2020-08-08:
@dennisguse asked here https://github.com/OpenTracksApp/OpenTracks/pull/362#issuecomment-670943430 if also the start and pause buttons should be "protected" of accidental clicks? 
I have the feeling that only the "stop" button is problematic. What do you think?

dennisguse, 2020-08-11:
About the pause button: I would also introduce the same behavior (as for proposed for stop).
In fact both buttons are almost doing the same except that the pause button keeps the GPS started and it does not move the TrackRecordedActivity.
So, from a data recording perspective both are very similar: basically if trigger any of them you record anymore.

pstorch, 2020-08-11:
Then we would have a consistent button behavior. Should the start behave the same?

dennisguse, 2020-08-11:
About start: I am still not sure if this is really needed.
Why not roll out pause/stop and check how it feels?

pstorch, 2020-08-11:
I've updated the PR accordingly.

commit 18546fd3 (2020-08-15): fix #241 stop button lock (#362)

* fix #241 stop button lock

Adds a delay and animation to the stop and pause buttons

- short vibration when starting to push the buttons
- long vibration when action was executed
- Toast to show "Hold to stop/pause recording"
closed 2020-08-15 by pstorch