OpenTracksApp/OpenTracks #244: Cumulative statistics

slecache, 2020-06-02:
**Is your feature request related to a problem? Please describe.**
There was "Cumulative statistics" in MyTracks menu that it seems be missing in OpenTracks.

**Describe the solution you'd like**
Display the same screen than one track "Statistics" but by cumulating the data of all the tracks.

dennisguse, 2020-06-02:
This was removed in one of my code cleanups.
The feature did not seem to be really useful (e.g., lack of filtering etc, no visualization of time) and made the code base slightly more complex (nowadays it would probably ok).
If you have time to work on this, it would be great.
#112 keeps track of a general statistics feature.

closed 2020-06-02 by dennisguse

slecache, 2020-06-03:
You're right: it's not really useful without filtering.

BTW, sorry for duplicate. I didn't find https://github.com/OpenTracksApp/OpenTracks/issues/112.