OpenTracksApp/OpenTracks #245: Import KMZ/KML from My Maps

slecache, 2020-06-02:
**Describe the bug**
When I try to import a KMZ/KML file generated by `My Maps` (previously exported from `MyTracks`), I got an error.
FYI, if I try to import a KMZ/KML file exported from `MyTracks` (without using `My Maps`), there is no error and the track file is well imported in `OpenTracks`.

**To Reproduce**

1. Export a track from `MyTracks` to `My Maps`.
2. Export this track from `My Maps` to KMZ/KML file.
3. Import this KMZ/KML file to `OpenTracks`.
4. `OpenTracks` try to import file but say it encountered an error.

**Technical information**
 - Device: [OnePlus 6T]
 - OS: [/e/ 0.9]
 - Version [v3.7.3]

dennisguse, 2020-06-02:
@slecache Could you provide us with a non-working KML/KMZ?

slecache, 2020-06-03:
You can find [here](https://github.com/OpenTracksApp/OpenTracks/files/4723025/tracks.zip) 2 samples (with MyTracks & MyMaps KMZ files each time):
- 2013/06/13: contains images in sub-folder and description in KML
- 2017/10/05: external images and no description in KML

I saw that MyTracks used /gx:MultiTrack/gx:Track when My Maps use /LineString/coordinates.

dennisguse, 2020-06-04:
Yes, LineString is not (yet) supported.

https://github.com/OpenTracksApp/OpenTracks/blob/master/src/main/java/de/dennisguse/opentracks/io/file/importer/KmlFileTrackImporter.java

@slecache If you have time to add it, it would be awesome!

slecache, 2020-06-05:
It's been a long time, I developed an Android application. Unfortunately, I no longer have a development environment. If I ever find some free time, I would take a look at it. But to be frank, you cannot count on me too much. Sorry...

BTW, you did a great job! Thanks a lot!

dennisguse, 2020-06-05:
@slecache Jupp, time is critical resources (especially in the summer).
Hope this issue is not affecting you too badly.

dennisguse, 2020-07-31:
I just looked into the provided files and LineString has one limitation: there are no timestamps for the provided coordinates.
It is just lat, long, and elevation.
So, if we import these we cannot compute the speed as well as the duration of the whole trip.

```
        <LineString>
          <tessellate>1</tessellate>
          <coordinates>
            1.099661,49.474236,0
            1.099203,49.474095,0
            1.09891,49.474111,0
            1.098025,49.474073,0
...
```

slecache, 2020-08-10:
Bad news for all my backuped tracks on My Maps... Anyway, it will be interesting to be able to import them.

BTW, isn't there a synthesis with the total / traveling time, the average / maximum speed, etc?

dennisguse, 2020-08-16:
Parsing the time info from the textual (localized) textual description is no fun.
What about an R script that does the data conversion from LineString offline?

dennisguse, 2020-09-28:
I guess handling LineString needs to be done somewhere else (due to the missing timestamp) before OpenTracks.

closed 2020-09-28 by dennisguse