OpenTracksApp/OpenTracks #247: Milliseconds are always 0

pe7ro, 2020-06-05:
Hi,
Currently I always get time without milliseconds on every device I tested. Seems like [this issue](https://stackoverflow.com/questions/23958640/location-gettime-always-returns-timestamp-without-milliseconds). Using `System.currentTimeMillis` in `TrackRecordingService.fillWithSensorDataSet` fixes the problem. IMO having an option in settings or making it a default would help me and others, who need more precise tracking (for example, speed measurement should improve much).

PS. Thanks for a cool app.

dennisguse, 2020-06-05:
@pe7ro What device are you using?

And why do you need millisecond precision?
I mean it is only GPS (i.e., not millimeter precise anyhow).
Do you have a specific use case in mind?

pe7ro, 2020-06-05:
I use bunch of Xiaomis, but AFAIK it is the idea of the GPS time - no milliseconds. I try to find out an activity type from the track, and knowing an exact speed is quite important. When the tracker logs position every second (the default setting for an app), then it is very rough. Also sometimes there can be 2 track points inside of one second, which leads to the speed being infinity.

dennisguse, 2020-06-05:
i still don't know what you want to do exactly.

Anyhow, I guess you could actually use the data from `Location.getSpeed()` and even combining it with `Location.getSpeedAccuracyMetersPerSecond()`.
I assume that the GPS hardwarehas some special magic how it determine actual position and speed while speed might not be just the diff between two positions divided by time.
Why don't you use the speed estimate directly?

https://developer.android.com/reference/android/location/Location#getSpeed()

closed 2020-06-13 by dennisguse