OpenTracksApp/OpenTracks #250: Unable to share track via bluetooth - bluetooth app crashes

mmacura311, 2020-06-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
1. open any track record
2. click "share" button
3. select "bluetooth" from options "share in app"
4. select bluetooth device to share with
5. Bluetooth app crashes

**Technical information**
 - Device: Nokia 5.1
 - OS: android 8.0.0
 - Version [e.g. v3.2.4]

I know it looks like the bluetooth app problem (and maybe is). But I can workaround this by saving the kml/gpx first to the phone as a file and share the file from file browser. Sharing files via bluetooth works for me, I'm able to share files or photos from different apps, but not directly from OpenTracks. I need to save as a file and then via bluetotooth or send it via mail.
I tried sharing the files with linux (fedora) laptop and also with windows laptop. It fails consistently for me.

dennisguse, 2020-06-06:
Can you provide a logcat output?

dennisguse, 2020-06-13:
Please reopen, when you can provide more details.

closed 2020-06-13 by dennisguse