OpenTracksApp/OpenTracks #251: Discussion: preference transmission to dashboards

dennisguse, 2020-06-06:
We just implemented the transfer of preferences from OpenTracks to OSMDashboard via Intent extras.
This is a rather nice solution (very simple) and therefore awesome :)

There is just one limitation: the intent will not be updated.
So, if a user opens OSMDashboard and subsequently in OpenTracks changes a preference, then this change will be ignored until OSMDashboard is started again.
I know it is a edge case, but anyhow.

What about a solution of using a ContentProvider that provides read access to OpenTracks sharedpreferences (incl. updates).
Then OSMDashboard could just open the URL similar to the data transfers and that's about it.

@pstorch @rgmf What are your opinions?

pstorch, 2020-06-06:
It updates everytime you open a new track from OpenTracks. 
The case you are referring to would be rather complicated I think: start live recording, open dashboard, go back to opentracks via task switcher, change settings, then again switch task to dashboard.
I think a user would instead go back from dashbard to opentracks via back button, change settings and open the dashboard again. Then settings are updated.

But maybe I'm wrong. Different users have different ways of working.

dennisguse, 2020-06-07:
I tend to have both apps running in parallel and switching depending on what I want to see.

pstorch, 2020-06-07:
Then you also change settings in between? Na, I don't want to argue. 
Let's share the preferences via ContentProvider with instant updates to be on the safe side.

rgmf, 2020-06-07:
At this moment the solution is great. The perfect solution would be a ContentProvider, of course, but like you say @dennisguse  it's an edge case. With time this could be coded.

By the way, congrats @pstorch for so quickly and awesome solution.

dennisguse, 2021-07-29:
It just works - no need to change anything :)

closed 2021-07-29 by dennisguse