OpenTracksApp/OpenTracks #276: Refactor trackId to a `class` instead of a `long`

dennisguse, 2020-06-12:
We might need to change trackId from `long` to something more useful like `UUID` to enable synchronization between multiple installtions of OpenTracks.

A first step would be to change trackId (which is used everywhere) to a class containing one long.

dennisguse, 2020-06-20:
After thinking about this, it might be great for maintenance while we should not replace the current long-trackId with a UUID-trackId.
The long-trackId is anyhow in the database (and a really useful primary key).

dennisguse, 2020-06-20:
#294 takes care about adding a global unique identifier.

dennisguse, 2020-06-20:
Blocked by #293.

pstorch, 2020-06-20:
> After thinking about this, it might be great for maintenance while we should not replace the current long-trackId with a UUID-trackId.
> The long-trackId is anyhow in the database (and a really useful primary key).

And used in the Dashboard interface.

dennisguse, 2020-06-20:
@pstorch Very valid point, so we should definitely not change the current trackId-column :)

dennisguse, 2020-08-18:
@pstorch @rgmf I plan to work on this in the evening.
AND it will affect a lot of places - so we almost definitely will get merge conflicts.
"But" it is just internal refactoring, so test will show if I break something.

dennisguse, 2020-08-18:
Arghhhh - I just spent 3 hours of copy and paste.
Result is here: https://github.com/OpenTracksApp/OpenTracks/tree/id%23276

Some tests (about 1/3) are still failing and I will check this tomorrow.
Those are mainly NPE.

And one lesson learned: we should work on our (existing) test.

commit 3c23f323 (2020-08-19): Implement Track.Id and Waypoint.Id.
Fixes #276.
commit 1ad91fc0 (2020-08-19): Implement Track.Id and Waypoint.Id (without tests).
Fixes #276.
commit aabe137c (2020-08-19): Implement Track.Id and Waypoint.Id.
Fixes #276.