OpenTracksApp/OpenTracks #278: Cadence sensor : allow tire circumference parameter instead of just a limited number of wheel size

serpico, 2020-06-13:
**Is your feature request related to a problem? Please describe.**
The exact tire dimension of the user's bike might not match the limited number (4) of wheel size proposed in parameter>Bluetooth>Wheelsize, 

**Describe the solution you'd like**
Add a field so user can enter the exact tire circumference in millimeters for his/her tire dimension, 

**Describe alternatives you've considered**
It's impractical to offer dozens of pre selected ERTO dimension covering all the cases, seems better to offer the most common as it is and add a "custom/tire circumference" field for the other.

**Additional context**
When looking up "bike tire circumference chart" user can find tables covering dozens of tire's dimensions and their corresponding circumference

dennisguse, 2020-06-13:
I partly agree we need more options for this (I just added the common race bikes).
I would rather go for a larger number of presets (even presenting milimeters in addition).
But entering milimeters - I measured it some ages ago, but nowadays would prefer presets.
Another thing is that we need to support metric and imperial values.

dr-m, 2020-06-24:
For the record, https://github.com/bailuk/AAT/ auto-detects the wheel size based on GPS and wheel rotation sensor measurements. If no wheel rotation sensor is configured or no pulses are being sent, it would display "Speed GPS"; otherwise, it displays "Speed S". Apparently the auto-detection takes some dozen wheel rotations on a straight road.

Similar auto-detection was also implemented in the Garmin Edge 705 bicycle computer more than 10 years ago. On that model, it usually took a few hundred meters of riding before a message popped up, indicating that the wheel circumference has been determined.

dennisguse, 2020-06-24:
@dr-m I like the idea to show the actual data source.
I will check, how we can do this! #302 for further reference.

I also like the idea to automatically detect the wheel circumference.
If somebody wants to look into this, it would be great.
I mean we can also talk to AAT and check how they are doing this.
No need to reinvent the wheel :)

edekeijzer, 2020-08-04:
> I partly agree we need more options for this (I just added the common race bikes).
> I would rather go for a larger number of presets (even presenting milimeters in addition).
> But entering milimeters [...]

@dennisguse if you prefer presets, would it be an idea to have a 'custom' option which shows an option to enter circumference manually? I am a mountainbiker myself and there are three ETRTO diameters (559, 584 and 622) and widths between 50 and 80 mm. (or even wider if you take fatbikes into account) which would result in hundreds of unique sizes. Next to that, the lower tire pressures on a mountain bike can influence the effective circumference, so measuring the length of a full rotation would be the most accurate option.
Strava has been neglecting us mountainbikers for years, having 'cycling' as only option (in Dutch it even says 'road cycling') while on foot I can go walking, hiking or running and in a boat I can either go rowing, canooing or kayaking. Please, care about us, don't be like Strava.

dennisguse, 2020-08-04:
Then let's go for a quick win here: replace the presets with the "enter your circumference"-option.
Would mm be ok? (That's what we use internally)
Only not so awesome thing is our support for imperial units.
Do you know what we should use then?

https://www.cyclinguk.org/cyclists-library/components/wheels-tyres/tyre-sizes

@edekeijzer "Don't be like...": YMMD

dennisguse, 2020-08-07:
@edekeijzer @serpico I changed the input to a textfield:
Can you give it a try if this works for you?
But it only accepts mm, nothing else ;)

https://github.com/OpenTracksApp/OpenTracks/tree/wheelsize%23278

dennisguse, 2020-08-15:
@edekeijzer @serpico Is this want you want?

PR #380 (2020-08-16): Wheel circumference is now a textfield accepting input as mm.
Fixes #278.
commit c8bdf8e6 (2020-08-16): Wheel circumference is now a textfield accepting input as mm.
Fixes #278.
closed 2020-08-16 by dennisguse