OpenTracksApp/OpenTracks #279: Record Pause/Resume and Stop buttons on LockScreen

pstorch, 2020-06-14:
What do you think of having buttons for Pause/Resume and Stop a recording on the LockScreen notification? Like most media players have.

Related functions are the "Keep screen on" and "Show statistics on lockscreen". But these are too invasive and change/prevent the lockscreen in total.

A little concern: if the phone is in my pocket while running, can the buttons be pushed accidentially? But during biking, if you have the phone mounted to the bike, such buttons would be comfortable.

dennisguse, 2020-06-14:
Nice idea!
What about start, pause/continue?
Then we have only one action that be selected or do you think stop is that important?

This might also allow interactions from a smartwatch that shows the notification.
But I don't have any for testing ;)

pstorch, 2020-06-14:
Pause/Resume would be the main case. For Stop it would be ok to unlock.

But Start? Without an already started recording there is no Notification active.

dennisguse, 2020-06-14:
When the GPS is started, there is a notification while not recording a track (yet).

pstorch, 2020-06-14:
Ahh, haven't used this feature yet 😁

pstorch, 2020-06-23:
I was looking into it a little bit. 

There is a possibility to add actions to a Notification:
https://developer.android.com/training/notify-user/build-notification#Actions

But these Actions are collapsed and only shown, when the user expands the Notification and not possible (at least I haven't found it) on the lockscreen. 

For Media Players there is a special style for Notifications:
https://developer.android.com/training/notify-user/expanded#media-style

These are the buttons I intended to use and those are visible and usable on the lockscreen. But i'm not sure if it is ok for a location tracking app to pretend to be a media player.

If time permits I'll to some experiments with it.

dennisguse, 2020-06-23:
We can also create a custom layout: https://developer.android.com/training/notify-user/custom-notification

pstorch, 2020-07-26:
I doubt that this feature is useful anymore. On my new phone the notifications on the Lockscreen are all collapsed. So even the notification of a music player app.  I still can expand the notifications, first clicking on them and after that I have to swipe down to see the player controls.
For me this doesn't make sense anymore. It's easier to unlock the phone with the fingerprint and control the app then.
So from my perspective, I would like to close this ticket.

dennisguse, 2020-07-26:
@pstorch works for me ;)

closed 2020-07-26 by dennisguse