OpenTracksApp/OpenTracks #281: Take a look at viewbinding and check if it saves some effort

dennisguse, 2020-06-16:
We have some viewbinding code that is not very complicated, but also not fun to take care of.
Automatic viewbinding would probably nice to make the UI changes slightly more fun.

https://developer.android.com/topic/libraries/view-binding

dennisguse, 2020-09-05:
@rgmf About the interval refactoring.
We can also try to refactor the TrackDataHub to a ViewModel (or even multiple ones).
We might be able to remove the internal interfaces.
Would this make sense?

rgmf, 2020-09-05:
> @rgmf About the interval refactoring.
> We can also try to refactor the TrackDataHub to a ViewModel (or even multiple ones).
> We might be able to remove the internal interfaces.
> Would this make sense?

Good question! I think it makes sense and the refactor can be worth it.

dennisguse, 2020-09-05:
https://github.com/OpenTracksApp/OpenTracks/issues/54
Is related - I guess.

dennisguse, 2020-10-31:
Done for fragments in #485.

Final step: activities.

dennisguse, 2021-03-19:
Already implemented.

closed 2021-03-19 by dennisguse