OpenTracksApp/OpenTracks #282: Request  "Always allow location access" on Android 10+

oksjd, 2020-06-19:
Phone: Oneplus 6
OS: LineagsOS 17.1
Version: v3.7.5
Source: F-Droid

Today I tried out this app on a bike ride. Has some very intuitive UI and helpful indicators.

One problem I encountered is that the system will block the location access of the app after awhile when the screen is locked and prompt for user confirmation. The option of always allowing location access is not present in permission settings, unlike other GPS tracking software such as OSMTracker (also available on F-Droid).

oksjd, 2020-06-19:
[image]
[image]

dennisguse, 2020-06-19:
Thanks for the report.
OpenTracks can only accesses the location sensor when the foreground service is running (i.e., the notification is shown) as only ACCESS_FINE_LOCATION is requested.

So, the lack of the ACCESS_BACKGROUND_LOCATION in the settings is actually correct.
This allows an app to _always_ access the location (even when not running, e.g., for geofencing).
And OpenTracks does do that.
Can you send us a screenshot of the blocking popup?

rgmf, 2020-06-19:
Maybe it's related with this: https://github.com/OpenTracksApp/OpenTracks/issues/177 ???

In #177 it was fixed disabling the battery saver for OpenTracks.

dennisguse, 2020-06-19:
I just checked the manifest of OSMTracker and the permissions are almost the same: https://github.com/labexp/osmtracker-android/blob/master/app/src/main/AndroidManifest.xml

So, now I am wondering why the permission dialog is different.
On my LineageOS 17.1, I get the same one permission request as @oksjd for OpenTracks (i.e. location access only while using the app).
And for OSMTracker (from F-Droid), I get a request incl. background access and while using the app.

Now I am curious why there is a difference in UI although the code/config looks very similar.

@oksjd What is the actual issue that you are experiencing?
The permission dialog opens again and again?

[image]

oksjd, 2020-06-19:
@rgmf The option has already set to "Not optimized" for OpenTracks in settings.

@dennisguse Unfortunately I haven't take a screenshot when the popup appeared yeasterday. It was something like "Allow bcakground location access for OpenTracks?" It appeared many times afterwards and I switched to OSMTracker for the rest of the journey.

Next time when I'm out for a ride I'll try to get the prompt displayed again.

dennisguse, 2020-07-05:
@oksjd Any more news on this issue?

wuistnedwissn, 2020-07-07:
I have the same problem. Opentracks will stop recording tracks after a little while when screen is off.
But I got no such popup messages.
I'm using a Huawei P20pro with Android 10 and EMUI 10.
Battery saving is set to off. But there's no chance to set the location allowance to always as this option isn't available as shown above.
Maybe this problem is linked to muted notfications?
If I remember right I set notifications to muted some days before I updated Android.
Or activating do not disturb causes this misbehaviour?
I'll give it a try.

wuistnedwissn, 2020-07-10:
I've tested several track recordings.
All show the same behaviour: nearly 2 minutes after closing der cellphone cover the recording stops.
So I think the only solution would be to change the setting for permanent access to location.

dennisguse, 2020-07-10:
@wuistnedwissn permanent (background) location access cannot be the solution for this issue.
Was your phone in power saving mode?
One question: when the recording stops: is there still a notification shown by OpenTracks?
If not the TrackRecordingService was killed by the operating system and that's it.
Nothing that OpenTracks can do about.

Even if we would enable background location access, we would need to start something then that actually records the location.
And we only have a foreground service (i.e., TrackRecordingService) and nothing else.
Moreover, background location access is really limited to only some locations per hour.
More info on this is available here: https://developer.android.com/about/versions/oreo/background-location-limits#apis

So, basically some Android distributors have some over optimization that introduces weird behavior :)

dennisguse, 2020-07-10:
This page has some infos about apps that are killed by phone manufacturer: https://dontkillmyapp.com/

wuistnedwissn, 2020-07-11:
@dennisguse yeah, the phone in power savings mode, but it was too under Android 9. And there were no problems.
The notification was still shown in notification bar.
On the other hand Fitotrack https://f-droid.org/de/packages/de.tadris.fitness does its job flawlwess.
Both apps have the same circumstances like power savings mode active, notifications are on.
Opentracks stops working after about 2 minutes, fitotrack works till I stop recording. 
Both apps came from the f-droid store.

dennisguse, 2020-07-12:
@wuistnedwissn I reproduce this behavior on Android 10 on a LineageOS.
While recording OpenTracks gets a WakeLock for the TrackRecordingService .
I am not sure what we can do in addition this if the power savings mode is started.
Could you check with FitoTrack how they handle this?

wuistnedwissn, 2020-07-13:
@dennisguse 
Sorry, I have no contact to the developer of FitoTrack, but Jannis has published FitoTrack under GPLv3.
You can find it under https://codeberg.org/jannis/FitoTrack

hchr, 2020-07-26:
> @dennisguse yeah, the phone in power savings mode, but it was too under Android 9. And there were no problems.
> The notification was still shown in notification bar.
> On the other hand Fitotrack [...]

I don't know if this might help to find the problem.
My setup is the same and I have no problems. Opentracks records fine in the background.
_However_ I do not only run Opentracks when I record. I always start OsmAnd+ first and start the GPX-logging.
Then the GPS is ready when I start the recording in Opentracks. I let both apps run in the background while recording

closed 2020-09-05 by dennisguse