OpenTracksApp/OpenTracks #294: For synchronization: generate random UUID for each track?

dennisguse, 2020-06-20:
For real synchronization (#90) and backup (#186), we need to be able to identify a track unambiguously.

For backup, we use generate a filename using the trackId (which is unambiguous per device, but not over several devices).
Here, we assume that the newest version version of track is always stored in OpenTracks and a new export/backup will overwrite prior backups.
Imports are handled very simple: for each imported track-file, a new track is created within OpenTracks.
Deduplication does not take place.

Long story short: for a real synchronization/backup solution, we need to be able to have a unique identifier per track that are stored _within_ the exported file (filenames tend to change over time).

So, what about adding a externalTrackId of type UUID?

Tasks:
* add column to track-table (must be non-null)
* write database migration script (incl. generation of UUID per existing track)
* handle export of externalTrackId to all GPX/KML
* handle import of pre-existing export without a unique UUID (incl. warning to the user about duplication)

iointerrupt, 2020-06-23:
Just wondering if **startime** column can be considered as a unique id of adding another column.  The chances of having two start at the exact same time down to the millisecond is unlikely.

pstorch, 2020-06-23:
Starttime could be problematic if you import tracks from other person's (your spouse or team members) to compare tracks for example. I'm in favour of a real uuid or guid column.

iointerrupt, 2020-06-23:
Didnt take that into account.  Agreed on real uuid.

dennisguse, 2020-06-23:
@pstorch @iointerrupt Is this a thumbs up for UUIDs?

If yes:
@rgmf Do you know how we could export the UUID nicely as GPX/KML (not part of the filename)? Should we add some custom field?

pstorch, 2020-06-23:
> @pstorch @iointerrupt Is this a thumbs up for UUIDs?

Yes 👍

rgmf, 2020-06-23:
> @rgmf Do you know how we could export the UUID nicely as GPX/KML (not part of the filename)? Should we add some custom field?

- In KML files: https://developers.google.com/kml/documentation/extendeddata#opaquedata We can use this for add an `<opentracks:uuid>` tag inside kml file, right?

- In GPX files: we have `<metadata>` tag with an `<extension>` one where we can add a `<opentracks:uuid>`.

In both cases we have to add a namespace. Can we do it? What URI for namespace can we use?

```xml
xmlns:opentracks="<namespace>"
```

Resources:
- https://developers.google.com/kml/documentation/extendeddata
- https://www.mapsmarker.com/kb/user-guide/how-to-use-gpx-extensions-to-customize-tracks/
- https://stackoverflow.com/questions/9195446/is-it-possible-to-add-your-own-tags-to-a-gpx-file-like-in-xml/9239579#9239579

commit c9869bab (2020-06-30): Generate UUID for each track (necessary for synchronization).
Fixes #294.
closed 2020-06-30 by dennisguse