OpenTracksApp/OpenTracks #295: Aggregated Statistics: add to PlantUML and message when empty list.

rgmf, 2020-06-20:
**Is your feature request related to a problem? Please describe.**
If there aren't tracks recorded then AggregatedStatisticsActivity layout shows an empty screen. It should show at least a message like this: "Record your first track to see aggregated statistics".

Also, in PlantUML doesn't appear AggregatedStatisticsActivity.

dennisguse, 2020-06-20:
Why not disable the menu item if there is nothing to be shown?

rgmf, 2020-06-20:
I've just done it and I think the result is good.

Disable the menu item means touch TrackListActivity instead of AggregatedStatisticsActivity.

Let me add a PR and then you take a decision ;)

dennisguse, 2020-06-20:
@rgmf If you like it, just do it ;)

closed 2020-06-20 by rgmf