OpenTracksApp/OpenTracks #303: Can't import .gpx files

secondlama, 2020-06-25:
Tried import some exported .gpx files from FitoTrack to OpenTracks, but get an error - "There were no files to import. Connect your device...".
Tried change filenames, folders - same error.

OpenTracks version 3.7.5 from F-Droid

dennisguse, 2020-06-25:
Can you check if the files have are named "someTrackName.gpx" - OpenTracks filters by file suffix.
You get this message when OpenTracks did not find any file within the selected folder.

rgmf, 2020-06-25:
@secondlama other thing you can to check is that `Settings --> Export/sharing file format` has the value: `GPX (without photos)`.

dennisguse, 2020-06-25:
@rgmf I just checked the code and you are right: I added this "feature" 4 weeks ago...
Should we change the message to incl. the selected file format or should we do import of all supported files within a folder?

rgmf, 2020-06-25:
@dennisguse I like the best the option of importing all supported files within a folder but the problem is that this approach can result in a potencial duplication.

dennisguse, 2020-06-25:
What about going with an error message until we added the UUID? I have some time on the weekend and can prepare the DB.

rgmf, 2020-06-25:
Great idea @dennisguse :clap: