OpenTracksApp/OpenTracks #305: First section of route missed off

fredx100, 2020-06-28:
The first section of a route is regularly missed off. I'm often missing the first 1km on a run. I've only noticed this recently (since around 19th April). I often am using the GPS in another app as I start my run (I plan my routes in OSMaps) and have confirmed my GPS connection is present (or at least it has my location).

I can't reliably reproduce but it is happening often. It's obvious from looking at the route by tapping the map icon at the top (I run circular routes and it's pretty obvious when something has gone wrong!).

What information would be useful to you? Issue #37 would be useful for diagnosis but I have tried to confirm gps lock by using another app.

 - Device: Sony Xperia X Compact
 - OS: Android: 9.0
 - Version:v3.7.5

dennisguse, 2020-06-29:
Thanks for the report - I have a similar issue at the moment as my phone's time to GPS fix is rather long (without A-GPS).
A workaround that might is actually start the recording (the notification shows the reported GPS accuracy then) and start your activity only after the accuracy is precise enough (less then 20m) - but then the time will be off.

Also it may give you a feeling how fast your GPS fix actually is (on my devices it varies a lot; something between 5s up to 60s).
We should also report this information in the notification when the GPS is started without starting a recording.

dennisguse, 2020-08-18:
With v3.9.0 that GPS can be started before starting a recording AND (that is actually new) the icon shows the actual GPS fix status.

fredx100, 2020-08-18:
Thanks for your responses. It does indeed appear to be a delay in getting a fix (at least, if I wait for opentracks to report a fix and then set off I can't seem to hit this issue). Having the icon display the appropriate status is a useful aid.

What more is reasonable to aim for for this enhancement? I'd love to demand that the GPS fix is always instant but I don't think this would be a helpful request! I only have v3.8.1 so can't comment on the the recent changes directly, but once the GPS status is clearly and reliably reported I would think that this enhancement has gone as far as it can.

dennisguse, 2020-08-19:
It will take some days until v3.9.0 gets released via Fdroid (we never know when exactlty; may take up to 7, days; and v3.9.0 was released only two days ago). When you get the update, it would be great to get your feedback on the new functionality.

fredx100, 2020-09-27:
I've finally had a look at the work and it looks good to me. Great stuff.

closed 2020-09-27 by fredx100

dennisguse, 2020-09-28:
Works was done by @rgmf :)

fredx100, 2020-09-29:
Thanks @rgmf !

rgmf, 2020-09-30:
@fredx100 you're welcome ;)