OpenTracksApp/OpenTracks #308: Shutdown on Huawei P10

geyr, 2020-07-02:
**The App finishes every track after max 9 minutes**
App ends up tracking after short time without any notification. (Short-)Track is saved. ???
**To Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

If applicable:
* add screenshots
* logcat output not available. ..?!!

**Technical information**
 - Device: Huawei P 10 VTR-L29
 - OS: EMUI 
 - Version 9.1.0

dennisguse, 2020-07-03:
Sounds like powersaving.
Can you check that the powersaving is disabled for OpenTracks on your device?
Otherwise, the powersaving mode will kill the TrackRecordingService (the notification will vanish) and the recording stops.

geyr, 2020-07-04:
Hi Dennis, thanks for your helpful hint: I looked under powermanagement and there was a restriction for OpenTracks. Unchecking the "background-running not allowed" did it!Thanks for the App, with regardsDierk--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

closed 2020-07-04 by dennisguse