OpenTracksApp/OpenTracks #311: Beurer AS99 not listed as heart rate sensor

ChristophWurst, 2020-07-06:
**Describe the bug**
I just purchased a little activity tracker that can send heart rate info via BLE. The model is a [Beurer AS99](https://www.beurer.com/web/us/products/active/sport-and-activity/activity-sensors/as-99.php). As OpenTracks did not show the sensor in the settings, I checked with a BLE explorer as well as a BLE heart rate monitor app. They both can connect to the device and list the sensor ID or show the actual numbers. Here are some screenshots

[image]
[image]

Ref https://github.com/OpenTracksApp/OpenTracks/blob/main/src/main/java/de/dennisguse/opentracks/util/BluetoothUtils.java#L40
Ref https://github.com/OpenTracksApp/OpenTracks/issues/239

**To Reproduce**
0. Pair the activity tracker with your phone
1. Open OpenTracks
2. Go to the settings
3. Open heart rate sensor picker
4. See just *None* as available options

If applicable:
* add screenshots
* add logcat

(let me know if they are helpful here and I'll capture them)

**Technical information**
 - Device: Pixel 3a
 - OS: Android 10 stock
 - Version: v3.7.5

ChristophWurst, 2020-07-06:
My bad, somehow it showed up on my third attempt. I was possibly just too fast with my interactions. Is the list of sensors loaded async? If so I would suggest to maybe add some text to the end of the sensors list for that so users know it's beneficial to wait a few secs.

Thanks a lot in any case. This app works nicely :rocket:

closed 2020-07-06 by ChristophWurst

dennisguse, 2020-07-07:
The discovery is done async - let me check if we can add a spinner there somehow.