OpenTracksApp/OpenTracks #316: Concept: how to store pause/resume

dennisguse, 2020-07-12:
At the moment, we store the pause/resume and signal lost (aka max distance) events as TrackPoints with a specific latitude (100.0 and 200.0).
In case of pause/resume even with two TrackPoints.
On export these are used as separators for individual track segments.

However, we have trouble when exporting and importing data; especially as we try to re-create the same database values (main issue is time).
Also we may run into issues that _we_ think a coordinate is a specific one, but it was reported this way by a GPS unit (see #315 where lat: 180 was in the database).

I guess, we need a cooler concept for this kind of data:
1. How do we store this kind of data?
2. How do we export/import this?

Follow-up to #315.

pstorch, 2020-07-12:
When I read the KML spec correctly, a `<gx:MultiTrack>` can have several `<gx:Track>` elements which are used to split a tour into several segments when interrupted by a break (a.k.a. pause/resume).

dennisguse, 2020-07-12:
@pstorch The actual question is: how do we store this in OpenTracks?
Multitrack is already used for GPX/KML (I also just read the spec ><) - better late than never.

pstorch, 2020-07-12:
The "clean" solution would be to also model it as segments in the database. But introducing a 1:n relationship from Track to Segments and then another 1:n relationship to TrackPoints might break a lot.

What about adding a SegmentNumber column in the TrackPoints table? And increase the SegmentNumber on every pause/resume.

What exactly are the "Lost Signal" TrackPoints used for?

dennisguse, 2020-07-12:
@pstorch It is not really signal lost - it is actually a segment breaker.
When you receive a GPS coordinate that farther away than _max distance_ (see settings) then a Pause-TrackPoint is inserted.
I guess the actual info is for drawing the map to not a draw a straight for a part where actually no useful measurements were received.
It is also used to compute the distance.
I just called it signal lost as in case of cycling it usually means that the GPS is not really precise/available.

pstorch, 2020-07-12:
Pulling the discussion from https://github.com/OpenTracksApp/OpenTracks/pull/315#issuecomment-657267366 to here.

> Pause/Resume is restored, but the timestamps are off.
> 
> And even one more thing: all GPS-based TrackPoints use GPS time, but pause/resume use device time (also start and stop time of a track).

Why do we need the extra timestamps of the pause/resume TrackPoints? Is it much different to the last/first valid TrackPoint of a segment?

dennisguse, 2020-07-12:
I guess, we don't except that we (nowadays) sort TrackPoints by time. So, duplicate timestamps are a problem as it is the same as the first/last valid track point. I am not sure, but we might have this issue at the moment. 

I like the idea about introducing an additional column.

dennisguse, 2020-07-13:
Correction: TrackPoints have an id and are also sorted by this id.
Time is just assumed to be monotonically increasing (may even be duplicated).
Which should not happen on a millisecond resolution except the pause/resume-trackpoints after import.

pstorch, 2020-09-28:
Follow up from #315 I tried again with current developer version, now all of my 90 tracks from my old backup are failing at import:

```
2020-09-28 21:21:06.638 31677-876/de.dennisguse.opentracks.debug D/KmzTrackImporter: Unable to parse kml in kmz
2020-09-28 21:21:06.646 31677-31677/de.dennisguse.opentracks.debug E/DecorView: mWindow.mActivityCurrentConfig is null
2020-09-28 21:21:07.192 31677-876/de.dennisguse.opentracks.debug E/AbstractFileTrackImporter: Track id is invalid.
2020-09-28 21:21:07.194 31677-876/de.dennisguse.opentracks.debug E/AbstractFileTrackImporter: Unable to import file
    org.xml.sax.SAXException: Parsing error at line: 69 column: 0. Invalid location detected: time=0: lat=100.0 lng=180.0 acc=0.0
        at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.addMarker(AbstractFileTrackImporter.java:312)
        at de.dennisguse.opentracks.io.file.importer.KmlFileTrackImporter.onMarkerEnd(KmlFileTrackImporter.java:200)
        at de.dennisguse.opentracks.io.file.importer.KmlFileTrackImporter.endElement(KmlFileTrackImporter.java:115)
        at org.apache.harmony.xml.ExpatParser.endElement(ExpatParser.java:164)
        at org.apache.harmony.xml.ExpatParser.appendBytes(Native Method)
        at org.apache.harmony.xml.ExpatParser.parseFragment(ExpatParser.java:523)
        at org.apache.harmony.xml.ExpatParser.parseDocument(ExpatParser.java:484)
        at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:322)
        at org.apache.harmony.xml.ExpatReader.parse(ExpatReader.java:285)
        at javax.xml.parsers.SAXParser.parse(SAXParser.java:390)
        at javax.xml.parsers.SAXParser.parse(SAXParser.java:187)
        at de.dennisguse.opentracks.io.file.importer.AbstractFileTrackImporter.importFile(AbstractFileTrackImporter.java:134)
        at de.dennisguse.opentracks.io.file.importer.KmlFileTrackImporter.importFile(KmlFileTrackImporter.java:38)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.parseKml(KmzTrackImporter.java:249)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.findAndParseKmlFile(KmzTrackImporter.java:177)
        at de.dennisguse.opentracks.io.file.importer.KmzTrackImporter.importFile(KmzTrackImporter.java:77)
        at de.dennisguse.opentracks.fragments.ImportProgressDialogFragment$ImportThread.importFile(ImportProgressDialogFragment.java:217)
        at de.dennisguse.opentracks.fragments.ImportProgressDialogFragment$ImportThread.run(ImportProgressDialogFragment.java:188)
2020-09-28 21:21:07.215 31677-31677/de.dennisguse.opentracks.debug E/DecorView: mWindow.mActivityCurrentConfig is null
```

dennisguse, 2020-09-28:
@pstorch Let's fix this one for real - basically removing these special trackpoints completely.
Proposal:
1. We add a segment id into `TrackPoint` (assuming order by id and the segment increases monotonically).
2. Convert existing data (within DB) to the new format.
(will be some be some SQL magic)

On import/export, we need to account for this new format.
Invalid trackpoints (everything that is weird) needs then to be ignored.

dennisguse, 2020-09-29:
After letting it sink in, I am not sure that this (segment-based) approach will work (especially considering how the statistics are computed):
* how do we compute the elapsed time if the user presses pause?
* how do we compute the average speed?

Computing the time is especially problematic if the user stationary (and therefore no new TrackPoints are stored), if he then presses pause, we will need to store something to mark the _correct_ time - otherwise, we will store something that is not true.
The same problem also occurs for resume/stop...
I am not even sure, how we handle this currently :)

Are there other edge cases that I am missing?
Or am I misunderstanding something in general?

pstorch, 2020-09-30:
Instead of invalid coordinates we could add a trackpoint type column (regular/pause) and store the current position also for the pause trackpoints.
Old trackpoints with invalid coordinates from backup could be converted accordingly to a pause trackpoint with the coordinates of the last trackpoint.

dennisguse, 2021-02-13:
Changes are implemented and were released in v3.15.0; implemented in #553.

closed 2021-02-13 by dennisguse