OpenTracksApp/OpenTracks #318: Long distances randomly divided in smaller parts and big part of the total path lost

vinceecniv87, 2020-07-13:
First of all, thanks a lot for developing and mantaining OpenTracks. I like the idea of opensource apps, and data ownership given to the final user :)

**Describe the bug**
Everytime I use the app while riding my bike, "random" start and stop events split the recording of a long path into very short distances (few hundreds meters) and somehow a big part of the total path is lost. This weekend I did a test comparing the app with other ones (DirtTracks, MyTracks, very similar but less complete than OpenTracks) while travelling by bus (in order to check in realtime what was happening in the 3 apps).  I could actually verify that while the other apps did work fine and register the whole path of 34Km, OpenTracks did only register 3 paths of 5.52 Km, 0.61Km and 0.52Km. Therefore, a part of the path was been lost and the remaining was split in 3 separate segments.

**To Reproduce**
Try to record a long path while keeping the screen of your phone locked. I didn't check what happens by keeping the screen on (as I am normally riding a bike when I use the app). Therfore I don't know if the problem is related to when the app runs in the background.

If applicable:
I add screenshots of the paths registered in OpenTracks and in the other apps. 
[image]
[image]
[image]
[image]
[image]
[image]

**Technical information**
 - Device: Redmi Note 5
 - OS: Android 
 - Version: 9 (Xiaomi MIUI Global 11.03 Stable

dennisguse, 2020-07-13:
Thanks!

Some questions:
* Why did you create three tracks?
  Was OpenTracks stopping the recording?
* How good was your GPS accuracy (shown in the notification)?
  Was it above the GPS accuracy threshold (see settings?)?
  Basically, OpenTracks ignores all GPS locations then...
* What is the settings of "Max distance"?
  If a new GPS location is farther away from the previous location a new segment is started.
* How do these Tracks look on a map (e.g., use OSMDashboard)?

vinceecniv87, 2020-07-13:
- I didn't do it on purpose. I started the recording process once and when I discovered openTracks had stopped, I started the recording again. However, I remember sometimes not doing anything, recording just once, and getting one or multiple very short tracks as a result, at the end of the ride.
- the accuracy floats below 10m in general, goes up to 22m sometimes and rarely above 50m. I have set the recommended value of 50m.
- max distance is set to 200m.
- I attach the screenshots of the maps
[image]
[image]
[image]
[image]
[image]
[image]

rgmf, 2020-07-14:
@vinceecniv87 maybe you have to set battery optimisation off for OpenTracks. If battery optimisation for OpenTracks is on then when OpenTracks goes to background then Android kill the service. Maybe this is the reason because you have these results with OpenTracks.

vinceecniv87, 2020-07-15:
Dear @rgmf,

what you suggested did work perfectly! I am a bit ashamed because I thought I had done it already (if I am not wrong a pop-up appears when using the app for the frist time). But still I am glad that we could solve the mistery very rapidly.

dennisguse, 2020-07-15:
@vinceecniv87 Thanks for the detailed report ;)

closed 2020-07-15 by dennisguse