OpenTracksApp/OpenTracks #340: Opentracks under contacts

gotros, 2020-07-29:
Hi,

under contacts some of my contact's mobilenumbers are falsely recognized as Opentrack coordinate (?). Thus I got the normal number to call and text and beneath I got the same number with the opentracks symbol next to it. When I try to open the phone number with Opentracks I get the following error.

[image]

I am using official LineageOS on my old S5 klte.

Thx in advance,
gotros

dennisguse, 2020-07-29:
@gotros 
Thanks!
I guess, I did not implement the intent-filter correctly to react to kml/kmz/gpx suffixes.

For devs: more info here: https://developer.android.com/guide/topics/manifest/data-element

dennisguse, 2020-07-29:
@gotros I guess the itnent-filters where far too optimistic (might have been matching everything).
Could you give it a try?

The following ZIP contains the fix (it is the PlayStore build; different package name than FDroid version).
[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/4997476/OpenTracksUpstream-releasePlayStore.zip)

gotros, 2020-07-29:
@dennisguse

Yes that fixed it. Thank you.

PR #341 (2020-07-29): Fixed too optimistic intent-filter for imports.
Fixes #340.
commit 699be450 (2020-07-29): Fixed too optimistic intent-filter for imports.
Fixes #340.
closed 2020-07-29 by gotros