OpenTracksApp/OpenTracks #346: Allow backup

pstorch, 2020-08-01:
**Is your feature request related to a problem? Please describe.**
I cannot perform an ADB backup of OpenTracks by myself, because the Manifest doesn't allow it.

**Describe the solution you'd like**
Allow backup, set 
`android:allowBackup="true"` in the manifest.

dennisguse, 2020-08-01:
This flag disables backup to Google Cloud (on Google-enabled phones).
I am not eager to send backups there.

https://developer.android.com/guide/topics/data/autobackup

dennisguse, 2020-08-01:
And we should add tools:replace="android:allowBackup" to enforce this for potential libraries that provide a AndroidManifest.xml

pstorch, 2020-08-01:
Ok, I understand. 
Can the cloud backup and the ADB backup be separated somehow? Because this flag also hinders me as a user to do a local backup on my own.

dennisguse, 2020-08-02:
I guess not - either you allow backups (incl. adb and Google Cloud) or not.
adb and Google Cloud Backup seem to use the same stuff on the device.

@pstorch Why would you like to backup the data via adb?
Is something missing in the KMZ export/import?
Or do we miss something else?

pstorch, 2020-08-02:
I tried https://codeberg.org/izzy/Adebar to do a complete Backup of my phone. But it gave me a list of apps which doesn't allow backups. So I came to open this ticket.
Yes, export via kmz works, but would be an extra manual step.

There are settings in the manifest to require client encryption of the backup or only allow device to device backup https://developer.android.com/guide/topics/data/autobackup#define-device-conditions . Would that be an option?

dennisguse, 2020-08-02:
It is API29 - but we can give it a try.
Should we create an issue for this?

pstorch, 2020-08-02:
> It is API29 - but we can give it a try.

API 28, but still very recent. When I read it correctly that means that there won't be any backup for older devices because they don't know about the `clientSideEncryption` requirement. I hope older devices really disallow backups if they can't fulfill this requirment.

> Should we create an issue for this?

If you don't want to tacke it here, I'm fine with a separate issue.

dennisguse, 2020-08-03:
We can also tackle it here ;)

dennisguse, 2020-09-11:
@pstorch Did you try `clientSideEncryption`?

pstorch, 2020-09-11:
No, haven't looked into it yet.

pstorch, 2020-09-14:
I learned in the past days how important trust is and now I think we should close the android backup way and concentrate on https://github.com/OpenTracksApp/OpenTracks/pull/383

closed 2020-09-14 by pstorch

hockeymikey, 2020-10-01:
> This flag disables backup to Google Cloud (on Google-enabled phones).
> I am not eager to send backups there.
> 
> https://developer.android.com/guide/topics/data/autobackup

Why?

dennisguse, 2020-10-01:
@hockeymikey There are people out there that are using phones with the original ROM (incl. Google Play Services) for different reasons.
Just from the top of my head: 
* phone does not support a custom ROM
* custom ROMs did not reach most recent Android version (e.g., Android 11)
* stability of the original ROM
* lack of practical knowledge
....

Therefore, we cannot deploy this functionality.
If you need this, we need create a separate build (which can be published also to F-Droid).

hockeymikey, 2020-10-01:
@dennisguse None of those matter at all. Let's go over the facts, you can disable syncing and google backups if you want to, if you're a pleb and running stock with gapps. What are you really preventing google from knowing if they had this app backed up? Your location? Nope, they are already getting all of that by your location requests to their servers using wifi and cell tower data. They are tracking your every step with the new covid tracker that is built into google services. If you have locations disabled, someone doesn't out there and they'll know and I'm sure it's "anonymous". They have complete control over the phone yet you're worried about backups for some reason? Give me a break and wake up. They scan your phone all the time with the play store, safety net is an obfuscated mess that no one knows what it really does or looks for. This is not a privacy feature at all, it's just a burden to the user.

mamoit, 2023-03-09:
@pstorch @hockeymikey
FYI: backups are now a thing :slightly_smiling_face: