OpenTracksApp/OpenTracks #347: To desugar or not to desugar?

dennisguse, 2020-08-03:
Recently, we enabled Java 8.

Does it bring a useful benefit to also enable desugaring?

@pstorch Any insight on this?

pstorch, 2020-08-03:
I thought it is mandatory when using Java 8 features.

dennisguse, 2020-08-03:
If I got it right, it is to use Java 8+ (at least certain features)?
Then just try it tomorrow: enable the preprocessor and check if Intellij provides some newer code optimizations :)

dennisguse, 2020-08-10:
I tested this last week and adding this library did not lead to any other optimizations suggested by Intellij.
So, I guess it is not required.

closed 2020-08-10 by dennisguse

dennisguse, 2020-08-21:
And I just figured out that, with desugar, we can use the newer date parsing as well.
https://github.com/OpenTracksApp/OpenTracks/commit/b2e8713db0b95ed50ed48f3f297f6f60e3786eb4

I should have enabled this earlier.
Mental note: AndroidStudio Analyze does not show such optimizations.

More infos of supported stuff: https://developer.android.com/studio/write/java8-support

@pstorch Thanks for making me aware of the desugar plugin!

commit af60e1fd (2020-08-20): Enable Google's desugar plugin (enables stream/map etc.).
Fixes  #347.