OpenTracksApp/OpenTracks #358: NPE (reported via Google Play)

dennisguse, 2020-08-06:
I don't know how to reproduce this.

```
java.lang.RuntimeException: 
  at android.app.ActivityThread.performLaunchActivity (ActivityThread.java:3304)
  at android.app.ActivityThread.handleLaunchActivity (ActivityThread.java:3443)
  at android.app.ActivityThread.handleRelaunchActivityInner (ActivityThread.java:5339)
  at android.app.ActivityThread.handleRelaunchActivity (ActivityThread.java:5247)
  at android.app.servertransaction.ActivityRelaunchItem.execute (ActivityRelaunchItem.java:69)
  at android.app.servertransaction.TransactionExecutor.executeCallbacks (TransactionExecutor.java:135)
  at android.app.servertransaction.TransactionExecutor.execute (TransactionExecutor.java:95)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:2040)
  at android.os.Handler.dispatchMessage (Handler.java:107)
  at android.os.Looper.loop (Looper.java:224)
  at android.app.ActivityThread.main (ActivityThread.java:7520)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:539)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:950)
Caused by: java.lang.NullPointerException: 
  at de.dennisguse.opentracks.MarkerEditActivity.onCreate (MarkerEditActivity.java:154)
  at android.app.Activity.performCreate (Activity.java:7893)
  at android.app.Activity.performCreate (Activity.java:7880)
  at android.app.Instrumentation.callActivityOnCreate (Instrumentation.java:1307)
  at android.app.ActivityThread.performLaunchActivity (ActivityThread.java:3279)
```

pstorch, 2020-08-07:
I could reproduce:

1. Record a track
2. Add Marker
3. Add Photo
4. Rotate device

dennisguse, 2020-08-07:
@pstorch: Thanks :)
I slowly start to dislike orientation change ><

dennisguse, 2020-08-15:
Fixed in ea97ff65dfd12324d4edf255455f571521635891.

closed 2020-08-15 by dennisguse