OpenTracksApp/OpenTracks #363: Sharing activities

slecache, 2020-08-10:
**Describe the solution you'd like**
I would like to share my activities in a photo containing some statistics. Could it be done with a custom dashboard like [OSMDashboard](https://github.com/OpenTracksApp/OSMDashboard)?

**Additional context**
Like this [one](https://support.strava.com/hc/article_attachments/360008726071/socialShare_run.jpg) see [here](https://support.strava.com/hc/en-us/articles/221089587-Sharing-your-Strava-Activities) for more details.

dennisguse, 2020-08-10:
Interesting idea!
If I got it right, you just want to extend the currently available sharing function that is implemented, right?

Just to clarify it:
1. you want to share just the image OR the image and the gpx/kml/kmz?
2. the image was taken by you while recording a track (or do we need some generic image)?

If we don't need Internet access to generate the image, we can do this within OpenTracks.

pstorch, 2020-08-11:
For the 1st example (other running apps do it similar), you would need to have some Track statistics (provided in the Dashboard interface), but you would also need to access the photo of a waypoint. We discussed this when adding the waypoints to the Dashboard interface. And there is no need for the Trackpoints. At the moment this sounds like a separate interface or a Feature withing OpenTracks (Share a Wapoint image and with added Track statistics).

slecache, 2020-08-11:
> If I got it right, you just want to extend the currently available sharing function that is implemented, right?

I though about a new CustomDashboard but extending the currently available sharing function sounds good to me.

> 1. you want to share just the image OR the image and the gpx/kml/kmz?
> 2. the image was taken by you while recording a track (or do we need some generic image)?

I would like to share a image generated from:
1. A photo of a waypoint
2. Some Track statistics : total duration, distance and speed average

> If we don't need Internet access to generate the image, we can do this within OpenTracks.

You're right!

hchr, 2020-08-11:
Would something like this be possible to?:
[image]

pstorch, 2020-08-11:
@hchr this would be something for the OSMDashboard as the map is involved.

dennisguse, 2021-03-19:
I was thinking about rendering a Android layout on top of an image.
This way the resulting overlay would be easily configurable.

commit 1fb195bd (2021-07-28): Share picture: if sharing a track, a share picture (PNG) is created in addition.

Fixes #363.
commit aa3f76c0 (2021-08-01): Share picture: if sharing a track, a share picture (PNG) is created in addition.

Fixes #363.
PR #888 (2021-08-01): Share picture: if sharing a track, a share picture (PNG) is created in addition
* Rendering is done via a layout (really easy to change).
* Rendering is done via ShareContentProvider (no tmp file needed)
* Background image is only OpenTracks logo for now

How to use: select a track and select share from the action menu.
Then the GPX/KML and a PNG is created.

For testing, I sent those via e-mail to myself :shrug: 

That's how it looks like:

[image]

Fixes #363.
commit fc0cec61 (2021-08-01): Share picture: if sharing a track, a share picture (PNG) is created in addition.

Fixes #363.
closed 2021-08-01 by dennisguse