OpenTracksApp/OpenTracks #364: SAXParser should check that input file is valid (xml schema)

dennisguse, 2020-08-10:
We do not seem to verify that a file to be imported is valid according to the XML Schema.
Should we do this?

https://docs.oracle.com/javase/tutorial/jaxp/sax/validation.html

PS: We may need to provide a custom XML schema for export/import the UUID.

dennisguse, 2020-08-10:
The XML schema introduced in: https://github.com/OpenTracksApp/OpenTracks/pull/312
also needs to used to verify the content.

And we need to bundle the XML schema files (as not Internet access is available in OpenTracks).

pstorch, 2020-08-11:
If you want to validate the XML schema do this with the absolute minimum which OpenTracks needs (be error tolerant).

dennisguse, 2020-08-11:
@pstorch Valid argument!
Then I would like at least to verify that the documents OpenTracks create adhere to the XML schema.
I need to check how we can configure the error resolution in the SAXParser - normally I would argue, we should only accept conforming documents.
But we already have at least one workaround for the timestamp parsing.

The only thing is that our parsing is not really error resilient as it does not account for correct nesting of elements (at least if I understood it right).
With XML schema verification, we get this basically for free.
Still we need to be careful about this!

dennisguse, 2022-05-08:
:-1: Android does not come with XSD support.
Although the API is available, the classes cannot be instantiated.

https://stackoverflow.com/questions/801144/android-schema-validation/6362749#6362749

:face_with_spiral_eyes:

closed 2022-05-08 by dennisguse