OpenTracksApp/OpenTracks #370: Avg. Pace: 4:60 instead of 5:00.

rgmf, 2020-08-12:
**Describe the bug**
I have a couple of tracks with an average pace of 4:60 but it would be 5:00.
Seem that there is a bug inside `getSpeedParts` in `StringUtils`.

**To Reproduce**
It's very difficult because you need to create a track with an average pace of exactly 5 minutes.

If applicable:
[image]

dennisguse, 2020-08-15:
Bug is in StringUtils.getSpeedParts() and this line:
```
return new Pair<>(String.format(Locale.US, "%d:%02d", minutes, seconds), unitString);
```

And it is a rounding issue - so, if you have 04:599 you get 04:60.

dennisguse, 2020-08-15:
I also added some test coverage for this function ;)

@rgmf nicely spotted!

commit 04f146c6 (2020-08-15): Bugfix: formatting of pace was not formatted correctly (e.g., 4:60).
Fixes #370.
closed 2020-08-15 by dennisguse