OpenTracksApp/OpenTracks #376: Implement UI to handle import conflicts

dennisguse, 2020-08-15:
Current implementation as of #312: either prevent import of already existing track (reported as error) or just duplicate the track (generate new UUID).

This can be improved by
* showing a dialog before start importing tracks (incl. number of tracks)
* showing a dialog for each conflicting file to be imported

Either way, we need the following options:
* skip (default)
* overwrite
* insert as new

PS: Follow-up to  #307 and #312.
https://github.com/OpenTracksApp/OpenTracks/pull/312#issuecomment-674363208
https://github.com/OpenTracksApp/OpenTracks/pull/312#issuecomment-674364972

dennisguse, 2020-11-23:
@rgmf Wasn't that implemented?

rgmf, 2020-11-23:
@dennisguse no, It wasn't. It could be the next thing to do after #518 , little by little. I can to take over this in the next days...

dennisguse, 2020-11-23:
My mistake: it is implemented for exports, not for imports. Was not aware of this...
But I guess it is not too urgent.

closed 2024-12-29 by dennisguse