OpenTracksApp/OpenTracks #379: implement functionality to analyse tracks into a separate application?

dennisguse, 2020-08-16:
Inspired by #377.

# Background
OSMDashboard is implemented as a separate application as it requires Internet access (which is not available within OpenTracks).
However, this really has the advantage that the two applications are decoupled and can evolve on there own (I also like that releases rarely depend on each other; as it makes my life easier).
So, we have an API in place to expose data (on user request) to other applications...

# What
Can we do the same for the data analytics part?
I mean, can we implement an "analytics dashboard"-application?

# Why
We have the two charts (code base is really not cool anymore) and only very limited functionality to actually get some information out of the tracks.
But: we will need a lot of _more_ code (and also libs) to make it really cool.
And for maintenance: it would be far easier to keep the whole system alive if the applications were to be separated - and also user could decide if they need this functionality.

@pstorch @rgmf @thrillfall

rgmf, 2020-08-16:
I like this proposal, it's like "doing one thing, well" :+1:

I'd vote for go ahead and create an "analytics dashboard"-application.

Although I have to say that my use case is record tracks with OpenTracks, upload them to NextCloud and download them to my laptop where I have tracks for years loaded in GoldenCheetah, so my main goal is spend time on OpenTracks. I could help when I have time with the new "analitycs dashboard"-application but my main goal is OpenTracks.

Finally, I like been able to see some aggregated stats on OpenTracks because is fast, but I understand that all that can be moved to "analitycs dahsboard"-application.

pstorch, 2020-08-17:
I also like the approach of splitting the app into individually manageable apps. We might encounter the issue that at some point in time we need to share some code/resources (e.g. activity icons). We already have some duplicates in the OSMDashboard. That's not necessarily bad, it helps us to decouple. But if this grows, we might have to think about a shared library (which has the drawback of buildtime dependency). Let's see how it turns out.
Also sharing the preferences might be more important, so coming back to #251

dennisguse, 2020-08-17:
Thanks for the input - then let's go down this rabbit hole ;)

My proposal for integration is actually the same way as OSMDashboard, we only have to figure out how the user may open the analytics application (but we need another request aka "show map" vs. "do data analytics").
We could it the same way as OSMDashboard, but just use another intent string.

My proposal is the following, we change the "analytics" button that we already have in TrackListActivity to open the analytics application (if installed).
As fallback, we can provide the overall stats page that was implemented by @rgmf.
For this, we can use the saem approach that we have for the OSMDashboard fallback (i.e., ShowOnMapProxyActivity).
We could also add the analytics button to the TrackRecordedActivity.

PS: For now I would leave the charts within OpenTracks, but I am really open to also remove them (if we have another application for this).

dennisguse, 2020-08-17:
And I forgot: what are we going to do about #377?
I would am in favor of moving in directly into the new application.

For #368: I would still be in favor to merge into OpenTracks as it will be simple enough.

@rgmf @thrillfall @pstorch Organizational stuff:
1. We need a name: ugly proposal AnalyticsDashboard (but I guess something O in front would be great, so icons are shown in the launcher together)?
2. Package name (won't be me this time ><)

rgmf, 2020-08-18:
> My proposal is the following, we change the "analytics" button that we already have in TrackListActivity to open the analytics application (if installed).
> As fallback, we can provide the overall stats page that was implemented by @rgmf.

Good for me.

> PS: For now I would leave the charts within OpenTracks, but I am really open to also remove them (if we have another application for this).

We need to define what analytic data is. I'd leave all data about every individual track in OpenTracks, so I'd leave speed/elevation charts in OpenTracks because is a track information like moving time, speed, average speed or gain, for example.

In this way, I'd consider analyitic data all data concerned with aggregated stats like:
- Number of ride, running... sesions: by week, by month, by year.
- Total km/mi by week, by month, by year.
- Your best avg speed/pace by week, by month, by year.
- Etc.

All this with diagrams/charts like @thrillfall  in #377

> For #368: I would still be in favor to merge into OpenTracks as it will be simple enough.

For runners this information is as necessary like time or distance. So, I think we should leave in OpenTracks. For analytic app I'd add the following: the best interval by week, by month, by year. This should be analytic data but the latter only a track data.

> We need a name: ugly proposal AnalyticsDashboard (but I guess something O in front would be great, so icons are shown in the launcher together)?

I like your ugly proposal :laughing: but I think about it.

> Package name (won't be me this time ><)

If there aren't other proposals I have several domains, one of them is rgmf.es, so It can be something like this: es.rgmf.analyticsdashboard.

dennisguse, 2020-08-20:
For naming:
Analytics, Overview, Summary, Graph/Graphs, Data, Calendar, Journal, Log, Visualize

And I got a request via mail for calorie estimation (might also fit in here).

rgmf, 2020-08-20:
Why not simply OpenTracksAnalytics??

dennisguse, 2020-09-12:
We got an additional feature request that fits into a separate analytics dashboard: https://github.com/OpenTracksApp/OpenTracks/issues/428

rgmf, 2020-09-13:
Another one could be https://github.com/OpenTracksApp/OpenTracks/issues/395

dennisguse, 2020-10-26:
#475 and #474 is related.

dennisguse, 2022-06-12:
I guess this discussion is outdated, right?
I mean we have the intervals tab, the aggregated stats, and the carts.
Also, people are actually exporting the data for evaluation.

closed 2022-06-12 by dennisguse