OpenTracksApp/OpenTracks #382: elevation and gps coordinates not shown

bkmystim, 2020-08-17:
Hi, on my samsung xcover4 there is currently no way to get the elevation and coordinates feature to work. In recorded tracks neither of them is shown, even when set in the settings menue.

Any idea? Thx BK

bkmystim, 2020-08-17:
App version 3.8.1 via fdroid

dennisguse, 2020-08-17:
@bkmystim These infos are only shown while recording.
They basically show the lat/long/elevation at your current position.
For this reason, these are not shown after a track was recorded.
Why would you need them (and which actually)?

bkmystim, 2020-08-17:
Would be great to know afterwards the meters you walked uphill and downhill, and to know lowest and highest altitude of the recorded track. Lat and long is actually not so important - afterwards.

dennisguse, 2020-08-17:
We are working on elevation gain/loss since some time and might have this ready soonish.
See #243.

closed 2020-08-17 by dennisguse