OpenTracksApp/OpenTracks #390: force close when stopping track

nutpantz, 2020-08-24:
**Describe the bug**
I start a track let it run for a few minutes and when i push to stop the track it force closes and automatically starts recording the same track again on restart... And is suck in the force close auto record loop
Tried force close and clear data and re install.  Same issue..
Works fine on my s7 with Google services
**To Reproduce**
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

If applicable:
* add screenshots
* add logcat output

**Technical information**
 - Device: Galaxy s4
 - OS:  LineageOS  androud 7.12
 - Version  v3.9.1

From fdroid no Google play or services installed

rgmf, 2020-08-24:
@nutpantz do you have the same issue when you pause the track? Your issue remember me an weird issue I had when new buttons feature were added. In my case the resource `ic_button_stop_anim.xml` was the problem, Android didn't found it but this was in its place. I fixed the problem uninstalling OpenTracks and installing it again.

But seem you cannot fix the problema with reinstallation :(

nutpantz, 2020-08-24:
Yes it appears it has the same issue when pausing the track..
(Which i have never done before)

pstorch, 2020-08-24:
That's strange. Are the Google services needed for the animation?
I don't have a device without, so unfortunately I can't reproduce.

nutpantz, 2020-08-24:
I can confirm after testing with my Nexus 7 that 3.8.1 was the last working version for this bug

pstorch, 2020-08-24:
3.9.0 introduced the long press + animation on the stop and pause button to prevent accidental stop of the recording.
Have you tried reinstalling the App as @rgmf mentioned? Be careful and backup your tracks first.

rgmf, 2020-08-24:
> That's strange. Are the Google services needed for the animation?
> I don't have a device without, so unfortunately I can't reproduce.

It works for me in phones with Google Services and without them.

nutpantz, 2020-08-24:
I have tried fresh install from fdroid to my Nexus 7 and reistalll to my s4 and Lenovo.. still same issue

dennisguse, 2020-08-24:
@rgmf Works for me (LineageOS 17.1; one S4/GT9505)

pstorch, 2020-08-24:
@nutpantz does it vibrate, when you push the stop/pause button?

Any chance to get access to the logcat?

nutpantz, 2020-08-24:
No vibrate just force close.. i can try for a log, but i don't have much hope

nutpantz, 2020-08-24:
From my Lenovo if it helps

---- 24 Aug 2020 9:21:55 a.m. ----

08-24 07:54:29.503 16920 16920 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 16920

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:145)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:129)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController.onStopTouch(TrackController.java:115)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController.lambda$new$1$TrackController(TrackController.java:89)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.-$$Lambda$TrackController$s7OiSHtai4u0cspUB6EmSrklKqA.onTouch(lambda)

08-24 07:54:31.949  2433  9053 I WindowState: WIN DEATH: Window{c62b02d u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackRecordingActivity}

08-24 07:54:31.952  2433  3667 I WindowState: WIN DEATH: Window{e0acdd u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity}

08-24 07:56:06.870 17039 17123 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 17039

08-24 07:56:06.870 17039 17123 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.run(TrackController.java:151)

08-24 07:57:43.679  2433  9056 I WindowState: WIN DEATH: Window{c7424c7 u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity}

---- 24 Aug 2020 9:21:55 a.m. ----

nutpantz, 2020-08-24:
Maybe better

---- 24 Aug 2020 9:21:55 a.m. ----

08-24 07:54:29.503 16920 16920 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 16920

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:145)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:129)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController.onStopTouch(TrackController.java:115)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController.lambda$new$1$TrackController(TrackController.java:89)

08-24 07:54:29.503 16920 16920 E AndroidRuntime: 	at de.dennisguse.opentracks.-$$Lambda$TrackController$s7OiSHtai4u0cspUB6EmSrklKqA.onTouch(lambda)

08-24 07:54:31.949  2433  9053 I WindowState: WIN DEATH: Window{c62b02d u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackRecordingActivity}

08-24 07:54:31.952  2433  3667 I WindowState: WIN DEATH: Window{e0acdd u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity}

08-24 07:56:06.870 17039 17123 E AndroidRuntime: Process: de.dennisguse.opentracks, PID: 17039

08-24 07:56:06.870 17039 17123 E AndroidRuntime: 	at de.dennisguse.opentracks.TrackController$ButtonDelay.run(TrackController.java:151)

08-24 07:57:43.679  2433  9056 I WindowState: WIN DEATH: Window{c7424c7 u0 de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity}

---- 24 Aug 2020 9:21:55 a.m. ----

rgmf, 2020-08-24:
I think the error is triggered when it tries to load the resource, so seem the error is the same I got. In my case Logcat said me that "thre resource ..... doesn't exists" but I went to the location and the resource existed. So I clean all and reinstall all and then It worked.

@nutpantz tomorrow I'll try to reproduce it.

pstorch, 2020-08-24:
Ok, that is the line, where the animated vector drawable is loaded:
https://github.com/OpenTracksApp/OpenTracks/blob/60d5e7e51510a2bf177375eb19c7ce24635e2569/src/main/java/de/dennisguse/opentracks/TrackController.java#L145

But I don't see in the log what error it is. 😕

rgmf, 2020-08-24:
@pstorch I'm confused... but I'll try to reproduce.

nutpantz, 2020-08-24:
Is this better?
---- 24 Aug 2020 10:34:47 a.m. ----

08-24 10:34:11.655  2433  2620 I am_create_activity: [0,21938115,2851,de.dennisguse.opentracks/.TrackListActivity,android.intent.action.MAIN,NULL,NULL,270532608]

08-24 10:34:11.663  2433  2620 I am_focused_activity: [0,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:11.701  2433  3667 I am_proc_start: [0,21844,10184,de.dennisguse.opentracks,activity,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:11.914  2433  2620 I am_proc_bound: [0,21844,de.dennisguse.opentracks]

08-24 10:34:11.918  2433  2620 I am_restart_activity: [0,21938115,2851,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:12.221 21844 21844 I am_on_resume_called: [0,de.dennisguse.opentracks.TrackListActivity]

08-24 10:34:12.631  2433  2484 I am_activity_launch_time: [0,21938115,de.dennisguse.opentracks/.TrackListActivity,953,953]

08-24 10:34:14.791  2433  9095 I am_create_activity: [0,209850111,2851,de.dennisguse.opentracks/.TrackRecordingActivity,NULL,NULL,NULL,335544320]

08-24 10:34:14.792  2433  9095 I am_pause_activity: [0,21938115,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:14.799  2433  9095 I am_focused_activity: [0,de.dennisguse.opentracks/.TrackRecordingActivity]

08-24 10:34:14.803 21844 21844 I am_on_paused_called: [0,de.dennisguse.opentracks.TrackListActivity]

08-24 10:34:14.805  2433  9053 I am_restart_activity: [0,209850111,2851,de.dennisguse.opentracks/.TrackRecordingActivity]

08-24 10:34:14.926 21844 21844 I am_on_resume_called: [0,de.dennisguse.opentracks.TrackRecordingActivity]

08-24 10:34:15.191  2433  2433 I notification_enqueue: [10184,21844,de.dennisguse.opentracks,123,NULL,0,Notification(pri=2 contentView=null vibrate=default sound=default defaults=0xffffffff flags=0xb color=0x00000000 category=service vis=PRIVATE),0]

08-24 10:34:15.192  2433  2433 I notification_alert: [0|de.dennisguse.opentracks|123|null|10184,0,0,1]

08-24 10:34:15.202  2433  2433 I notification_enqueue: [10184,21844,de.dennisguse.opentracks,123,NULL,0,Notification(pri=2 contentView=null vibrate=default sound=default defaults=0xffffffff flags=0xb color=0x00000000 category=service vis=PRIVATE),1]

08-24 10:34:15.203  2433  2433 I notification_alert: [0|de.dennisguse.opentracks|123|null|10184,0,0,1]

08-24 10:34:15.211  2433  2433 I notification_enqueue: [10184,21844,de.dennisguse.opentracks,123,NULL,0,Notification(pri=2 contentView=de.dennisguse.opentracks/0x1090086 vibrate=default sound=default defaults=0xffffffff flags=0x4b color=0x00000000 category=service vis=PRIVATE),1]

08-24 10:34:15.213  2433  2433 I notification_alert: [0|de.dennisguse.opentracks|123|null|10184,0,0,1]

08-24 10:34:15.310  2433  2484 I am_activity_launch_time: [0,209850111,de.dennisguse.opentracks/.TrackRecordingActivity,507,507]

08-24 10:34:15.322  2433  9095 I notification_expansion: [0|de.dennisguse.opentracks|123|null|10184,0,1,131,111,0]

08-24 10:34:15.390   250   250 I sf_frame_dur: [de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity,16,2,0,1,3,1,0]

08-24 10:34:15.833  2433  3483 I notification_visibility: [0|de.dennisguse.opentracks|123|null|10184,1,642,622,0,0]

08-24 10:34:21.178  2433  2624 I notification_visibility: [0|de.dennisguse.opentracks|123|null|10184,0,5987,5967,0,0]

08-24 10:34:22.420  2433  2477 I am_pss  : [21844,10184,de.dennisguse.opentracks,14326784,14326784]

08-24 10:34:22.562  2433  2468 I am_crash: [21844,0,de.dennisguse.opentracks,948452932,java.lang.ClassCastException,android.graphics.drawable.BitmapDrawable cannot be cast to android.graphics.drawable.VectorDrawable,AnimatedVectorDrawable.java,317]

08-24 10:34:22.563  2433  2468 I am_finish_activity: [0,209850111,2851,de.dennisguse.opentracks/.TrackRecordingActivity,force-crash]

08-24 10:34:22.568  2433  2468 I am_focused_activity: [0,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:22.568  2433  2468 I am_pause_activity: [0,209850111,de.dennisguse.opentracks/.TrackRecordingActivity]

08-24 10:34:23.072  2433  2478 I am_resume_activity: [0,21938115,2851,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:24.293  2433  2620 I am_proc_died: [0,21844,de.dennisguse.opentracks]

08-24 10:34:24.294  2433  2620 I am_schedule_service_restart: [0,de.dennisguse.opentracks/.services.TrackRecordingService,1000]

08-24 10:34:24.295  2433  2433 I notification_cancel: [1000,2433,de.dennisguse.opentracks,123,NULL,0,0,0,8,NULL]

08-24 10:34:24.296  2433  2433 I notification_canceled: [0|de.dennisguse.opentracks|123|null|10184,8,9105,9085,8463]
[... 33 lines cut ...]
08-24 10:34:25.347  2433  2484 I am_activity_launch_time: [0,21938115,de.dennisguse.opentracks/.TrackListActivity,1030,1030]

08-24 10:34:30.481  2433  3483 I notification_visibility: [0|de.dennisguse.opentracks|123|null|10184,0,5935,5532,0,0]

08-24 10:34:31.497  2433  9053 I am_crash: [21904,0,de.dennisguse.opentracks,948452932,java.lang.ClassCastException,android.graphics.drawable.BitmapDrawable cannot be cast to android.graphics.drawable.VectorDrawable,AnimatedVectorDrawable.java,317]

08-24 10:34:31.499  2433  9053 I am_finish_activity: [0,21938115,2851,de.dennisguse.opentracks/.TrackListActivity,force-crash]

08-24 10:34:31.532  2433  9053 I am_pause_activity: [0,21938115,de.dennisguse.opentracks/.TrackListActivity]

08-24 10:34:32.112  2433  3667 I am_destroy_activity: [0,21938115,2851,de.dennisguse.opentracks/.TrackListActivity,finish-imm]

[... cut ...]
08-24 10:34:33.817  2433  2433 I notification_cancel: [1000,2433,de.dennisguse.opentracks,123,NULL,0,0,0,8,NULL]

08-24 10:34:33.817  2433  2433 I notification_canceled: [0|de.dennisguse.opentracks|123|null|10184,8,9271,8868,8692]

08-24 10:34:33.818   250   250 I sf_frame_dur: [de.dennisguse.opentracks/de.dennisguse.opentracks.TrackListActivity,17,0,1,0,1,4,1]

08-24 10:34:33.835   250   250 I sf_frame_dur: [Application Error: de.dennisguse.opentracks,14,3,0,0,0,2,0]

---- 24 Aug 2020 10:34:47 a.m. ----

pstorch, 2020-08-24:
That could be a hint:
`java.lang.ClassCastException,android.graphics.drawable.BitmapDrawable cannot be cast to android.graphics.drawable.VectorDrawable,AnimatedVectorDrawable.java,317]`

pstorch, 2020-08-24:
I maybe found something regarding the compat library for animated vector drawables:

https://stackoverflow.com/questions/40103290/getdrawable-with-animatedvectordrawable-crashes-in-android-version-4

The mentioned devices are a bit older with Android 6 and 7. I also got an error in an Emulator with such an old version.

But when I use the compat library it fails because it cannot find a PNG fike for the drawable (maybe as fallback?).

rgmf, 2020-08-24:
@pstorch so can have the same issue  with GPS Status icon. Good hint.

pstorch, 2020-08-24:
Some more analysis:

On Android Emulator with API 21 (Android 6) and API 23, I get this Exception:

```
08-24 20:33:15.239 6416-6416/de.dennisguse.opentracks.debug E/AndroidRuntime: FATAL EXCEPTION: main
    Process: de.dennisguse.opentracks.debug, PID: 6416
    android.content.res.Resources$NotFoundException: File res/drawable/ic_button_stop_anim.xml from drawable resource ID #0x7f0800a4
        at android.content.res.Resources.loadDrawableForCookie(Resources.java:2428)
        at android.content.res.Resources.loadDrawable(Resources.java:2330)
        at android.content.res.Resources.getDrawable(Resources.java:758)
        at android.content.Context.getDrawable(Context.java:402)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:145)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:129)
        at de.dennisguse.opentracks.TrackController.onStopTouch(TrackController.java:115)
        at de.dennisguse.opentracks.TrackController.lambda$new$1$TrackController(TrackController.java:89)
        at de.dennisguse.opentracks.-$$Lambda$TrackController$s7OiSHtai4u0cspUB6EmSrklKqA.onTouch(lambda)
```

With an Emulator on API 25 (Android 7.1.1), I get this Exception:
```

2020-08-24 20:35:49.461 5079-5150/de.dennisguse.opentracks.debug E/AndroidRuntime: FATAL EXCEPTION: Thread-2
    Process: de.dennisguse.opentracks.debug, PID: 5079
    android.view.ViewRootImpl$CalledFromWrongThreadException: Only the original thread that created a view hierarchy can touch its views.
        at android.view.ViewRootImpl.checkThread(ViewRootImpl.java:6891)
        at android.view.ViewRootImpl.invalidateChildInParent(ViewRootImpl.java:1083)
        at android.view.ViewGroup.invalidateChild(ViewGroup.java:5205)
        at android.view.View.invalidateInternal(View.java:13656)
        at android.view.View.invalidate(View.java:13620)
        at android.view.View.invalidate(View.java:13604)
        at android.widget.ImageView.setImageDrawable(ImageView.java:531)
        at androidx.appcompat.widget.AppCompatImageButton.setImageDrawable(AppCompatImageButton.java:94)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.run(TrackController.java:151)
        at java.lang.Thread.run(Thread.java:761)
```

Now with the AnimatedVecotrDrawableCompat.

API 21 and API 23:
```

08-24 20:37:51.578 6601-6601/de.dennisguse.opentracks.debug E/MessageQueue-JNI: android.content.res.Resources$NotFoundException: File res/drawable-xxhdpi-v4/ic_button_stop.png from xml type xml resource ID #0x7f0800a3
        at android.content.res.Resources.loadXmlResourceParser(Resources.java:2645)
        at android.content.res.Resources.loadXmlResourceParser(Resources.java:2600)
        at android.content.res.Resources.getXml(Resources.java:1098)
        at androidx.vectordrawable.graphics.drawable.VectorDrawableCompat.create(VectorDrawableCompat.java:654)
        at androidx.vectordrawable.graphics.drawable.AnimatedVectorDrawableCompat.inflate(AnimatedVectorDrawableCompat.java:472)
        at androidx.vectordrawable.graphics.drawable.AnimatedVectorDrawableCompat.createFromXmlInner(AnimatedVectorDrawableCompat.java:254)
        at androidx.vectordrawable.graphics.drawable.AnimatedVectorDrawableCompat.create(AnimatedVectorDrawableCompat.java:234)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:147)
        at de.dennisguse.opentracks.TrackController$ButtonDelay.<init>(TrackController.java:131)
        at de.dennisguse.opentracks.TrackController.onStopTouch(TrackController.java:117)
        at de.dennisguse.opentracks.TrackController.lambda$new$1$TrackController(TrackController.java:91)
        at de.dennisguse.opentracks.-$$Lambda$TrackController$s7OiSHtai4u0cspUB6EmSrklKqA.onTouch(lambda)
```

On API 25, same as without compat library. 

The compat library is for API level before 24.

On API 27 it works without error.

pstorch, 2020-08-24:
The thread issue at API 25 I could fix.

pstorch, 2020-08-24:
> @pstorch so can have the same issue with GPS Status icon. Good hint.

Maybe

pstorch, 2020-08-24:
I read that we have to use `android.defaultConfig.vectorDrawables.useSupportLibrary = true` in the build.gradle file. 
But then I get a lot of other errors already at start of OpenTracks. Looks like then we have to use VectorDrawableCompat everywhere.

pstorch, 2020-08-25:
Maybe as a quick fix: remove the animation until we've found a proper way to run them on Android 6 (API 21).

rgmf, 2020-08-25:
Reviewing the the code about GPS Status animation what I did was that the animation GPS Status button only run if the instance was an instance of `AnimatedVectorDrawable`:

```java
if (startGpsMenuItem.getIcon() instanceof AnimatedVectorDrawable) {
    ((AnimatedVectorDrawable) startGpsMenuItem.getIcon()).start();
}
```

it's a lesser evil until we find a solution for this.

Did you find a solution @pstorch? I'm stuck with it.

Some interesting links that can explain why `useSupportLibrary = true` doesn't work and what can be the solution:
- https://stackoverflow.com/a/41408100/8727038
- https://stackoverflow.com/questions/36484748/appcompat-23-3-support-vectors-no-longer-work/36484749#36484749

pstorch, 2020-08-26:
Found it :tada:

closed 2020-08-27 by pstorch

rgmf, 2020-08-27:
Great!!!! :tada:

commit 24ff73ab (2020-08-27): Fix force close on stop button for older devices (#393)

* correct thread handling in ButtonDelay

Discovered in #390 for devices with API Level 25. Not fixing the whole issue, but a first step.

* fix #390 crash when stopping for older devices

The VectorDrawable ic_button_stop.xml referenced from withing the AnimatedVectorDrawable ic_button_stop_anim.xml was a problem for older devices. Now the VectorDrawable is expanded (duplicated) inside the AnimatedVectorDrawable. Now it is working. Even with the animation.