OpenTracksApp/OpenTracks #395: follow equipment usage

tclavier, 2020-08-30:
As a runner I want to know how many km I run with my shoes.
In order to do that, I want to add my equipment in OpenTracks and attach this equipment to tracks.
In equipment window, I want to see the usage statistics : distance and time usage.

dennisguse, 2020-08-30:
Just to be clear: you have different shoes that you are using for different training sessions (during the day/week/whatever) and you want to know how many km you did with them.
I guess, you want to know when you need to replace them?

In any case, you could assign different activity types (like hiking vs. running) to group different recordings together. Then you can use the aggregated statistics functionality that is reachable from the track list.

tclavier, 2020-08-31:
Yes, I have different pairs of shoes. I run an average of 30 km per week, so I change shoes an average of 3 times a year. I have one pair for road running and another one for trail running. And I buy my shoes on special offer with an option to return them within 30 days in case of dissatisfaction. So I have around 4 to 6 pairs of shoes that are worn out at the same time.

dennisguse, 2020-09-01:
Understood.

In general a good idea.
I mean not really a core feature, but yolo.

Only thing as usual: might take sooommee before (or never) be implemented.
@tclavier Can you live for now "misusing" the different activity icons?

tclavier, 2020-09-10:
@dennisguse for now, i duplicate my activities in "garmin connect" :-(

DavidHenryThoreau, 2023-11-09:
It'd be nice to add equipment like shoes brand, or bike maybe a blank tag that user can add and set to each outdoor actvities