OpenTracksApp/OpenTracks #396: Unable to stop!

mbnoimi, 2020-08-30:
**Describe the bug**
As show in the following gif OpenTrack unable to stop at all!
I tried to reinstall it but it still reproduce same bug.

[image]

[image]

**Technical information**
 - Device: ASUS Zenfone 6
 - OS: Android 5.0
- Version: 3.9.1

rgmf, 2020-08-30:
This bug is fixed... Soon there will be a new release. Thank you for open this Issue ;)

Here more info: https://github.com/OpenTracksApp/OpenTracks/issues/390

mbnoimi, 2020-08-30:
@rgmf Oops I didn't look in the closed issues

dennisguse, 2020-08-30:
There is another issue related that we are going to resolve (soonish): #388.
Once the TrackRecordingService dies while recording, the UI never gets this info and it requires resetting the preferences (which is not possible while the UI thinks a recording is ongoing).
So, the only option to recover from this state is to delete all app data.

dennisguse, 2020-08-30:
Fixes (and a workaround) are in v3.9.2 (also for the behavior in my previous comment: https://github.com/OpenTracksApp/OpenTracks/issues/396#issuecomment-683451712)

Let's leave this issue open until this release reaches F-Droid (will take up to 7 days...).

@mbnoimi It would be great if you could test the fixes.
Here is the build that I just pushed to the PlayStore:
[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/5147475/OpenTracksUpstream-releasePlayStore.zip)

mbnoimi, 2020-08-30:
@dennisguse I installed it and imported the old tracks it works fine but tomorrow I'll test it practically (I've 8 km to walk)

mbnoimi, 2020-08-31:
@dennisguse I can see you've released a new version https://github.com/OpenTracksApp/OpenTracks/releases/tag/v3.9.3 may you please upload the upstream?

dennisguse, 2020-08-31:
What do you mean?
I can only update the PlayStore as Fdroid uses its own key for signing.

mbnoimi, 2020-08-31:
> What do you mean?

I only want an apk because waiting 7 days is too much,

dennisguse, 2020-09-01:
1. The APKs are not the same (the application the is different)
2. I cannot build Fdroid (and I guess never will)
3. Creating a build takes me about 10min - and I rather waste my spare time differently
4. We are using the PlayStore build to finance our expenses

@mbnoimi If you have time and want to take a look on how to create nightly builds - feel free.
I guess some people might be interested in getting a nightly F-Droid repo.
One advantage: as app version, we use the #commits (main branch) - so this can be directly used for nightly builds.

dennisguse, 2020-09-01:
@mbnoimi I totally forgot: promo codes for the play store are available here: https://github.com/OpenTracksApp/OpenTracks/issues/97

FHenry, 2020-09-02:
Same bug here l, hope New apk on f-droid soon

rgmf, 2020-09-02:
> Same bug here l, hope New apk on f-droid soon
> 
> Edit: Finnaly pay from store, you deserve it for this opensource software

Version 3.9.2 is already in f-droid and this version should fix this bug ;)

dennisguse, 2020-09-02:
Saldy, export is not yet working (needs v3.9.3).

closed 2020-09-02 by dennisguse