OpenTracksApp/OpenTracks #403: gps is kept on when not recording

nutpantz, 2020-09-02:
Galaxy s4 LineageOS 712

After stopping a track the device keeps the icon for gps use in status bar and only disappears if i go into recents and swipe it away or if i force stop open tracks. The open tracks icon in the status bar works fine. 
If the gps keeps running out will kill the battery quickly.
For me it kept running for over two minutes before i force stopped it.
It appears to work fine on my s7 with Android 8

Start the app start a track, end the track and watch the gps active icon (usually a upside down teardrop poi marker icon).

rgmf, 2020-09-02:
I can reproduce it... :+1:

rgmf, 2020-09-03:
Catched and fixed. What an error!!!! :see_no_evil:

This becomes in my favourite screw up :sweat_smile:

closed 2020-09-03 by rgmf

dennisguse, 2020-09-03:
@rgmf Sounds like we need another release.
What do you want 3.9.4 or 3.10.0 (because the interval view is included).

rgmf, 2020-09-03:
@dennisguse I'm with this: https://github.com/OpenTracksApp/OpenTracks/issues/406

Do you have those issues. I'm fixing them now.

commit 36dc43f7 (2020-09-03): Bugfix: location handler didn't remove updates. Fixes #403.
commit b88f8172 (2020-09-03): Bugfix: location handler didn't remove updates. Fixes #403.