OpenTracksApp/OpenTracks #411: Opening gpx file from OT in OsmAnd~ crashes OsmAnd~ since last update to 3.9.3

Jakethethird, 2020-09-04:
**Describe the bug**
I have always opened the gpx files from the OT interface in OsmAnd and never had a problem. But after the last update to 3.9.3, it now starts OSMand which then crashes and closes after showing the initial OSMand start up screen. If I select an alternative mapping program (MMTracker), it works OK.

**To Reproduce**
1. Go to OT page showing all recorded routes
2. Click on button to open gpx file in another program
3. select OsmAnd~ from the programs suggested.
4. OsmAnd start page appears, then program is "stopped" and closes. Rebooting phone makes no difference.

**Technical information**
 - Device: motoG (2014)
 - OS: [e.g. LineageOS 14.1 with microG, no GApps]
 - Version [e.g. v3.9.3]

dennisguse, 2020-09-04:
Found the bug and it really hurt - we were sending data as intent extra and the data could not be unparceled.
But the actual fun fact: it was not even used...

PS: Cool that somebody is using this functionality!
I was recently wondering if we should keep it or not.
Why are not using OSM Dashboard?

closed 2020-09-04 by dennisguse

Jakethethird, 2020-09-05:
Thanks for fixing this so quickly. I always use this as the easiest way to get the .gpx file into osmand. When I first discovered this, I thought "wow, that is so cool"
Once moved across, I can access it via the osmand dashboard as it is in the list. But I only do this for new or interesting routes - most of my training is on just a few local loops (biking).

commit 38fadda7 (2020-09-04): Bugfix: show on map via GPX/KML/KMZ was failing.
We were sending the trackId as intent extra (actually unused) and this trackId could not be unparceled.
Fixes #411.