OpenTracksApp/OpenTracks #412: Get a rough Idea of the track that we have taken without using any other app

PhantomO2, 2020-09-04:
So, I am a big fan of this app. I really like the fact that, it doesn't need any unnecessary permission. Moreover, this app doesn't have "have full network access" and "receive data from internet" permissions. So, can you please provide a feature in this app through which we can see the approx track taken during an activity without having to open any other app with inbuilt map. Just imagine, opening a "GPX" file without any map on it (without using internet). Just the lines indicating the shape of the track formed during any activity. As this is a privacy friendly app and I am using it so that my data won't leak or gets uploaded to the cloud, and hence I thought that it would be great if this app can give me rough idea of the shape of the path I took during running. I am not saying that the shape should be accurate ( I guess for that you would have to integrate "have full network access" permission ). And please, never drop this privacy friendly approach for this app. Once again, I really appreciate your work for making this world a little bit more privacy friendly. Thanks, for your contributions.

PhantomO2, 2020-09-04:
[image]

nutpantz, 2020-09-04:
I'm confused, i thought osm dashboard does this? If you add the offline map you get the track on a map. 
https://f-droid.org/en/packages/de.storchp.opentracks.osmplugin/

PhantomO2, 2020-09-04:
Yep, but osm plugin uses "have full network access" permission. And I don't think that most of the people really check the source code to see, if the app is truly privacy friendly. That's why I like the apps that are on the playstore and that don't have unnecessary permissions like " have full network access",  "receive data from internet" etc. Btw since this is an app that checks all the box from my wish list, that's why I asked the Dev to add this feature in this app (No need for the internet, as I was just requesting for the shape created by the activity and not the map).

dennisguse, 2020-09-04:
I see the following options:
1. Create a variant of OSM Dashboard does not require network access permission.
   Then the user has download the maps manually.
   We could do it as some kind of build flavor.
2. Draw the graph within OpenTracks
  1. Just use an image view and do custom drawing (I have no clue, how difficult WGS84 projection is)
  2. use mapforge or another library just without a map as background (I am not a fan as this adds complexity and we have OSM Dashboard as map application already in place)

Limitations for within-OpenTracks: we do not yet use the compass sensor - so showing device orientation is not possible at the moment.

Usual disclaimer: personally, I don't need this feature, so I will not spend anytime soon on this (expect stupid comments for now).

@pstorch Any other options that you see?

pstorch, 2020-09-05:
@dennisguse that are pretty much our options. Don't see another one.

The OSM Dashboard doesn't access the Internet as long as the user doesn't give the consent to it. But of course the user needs to trust that feature.

nutpantz, 2020-09-07:
I would endorse osmdashboard without the internet permission.. I'll be honest I'm s privacy freak and i had osmdashboard blocked by my firewall for months and just looked at the track shape before i even bothered to RTFM and found out i could add a map.. i just wish it could use maps build it by "mobac"... But then it might i never did read much of the manual.
Maybe you could add a "use no internet switch" to osmdashboard now? Users who don't (or can't understand)look really have to trust the developer to be honest about everything and the paranoid (like me) use a firewall.

pstorch, 2020-09-08:
I've created a ticket for it on the OSMDashboard repository.

vanous, 2020-10-07:
Maybe milkha would work with OT export, not sure, i haven't tried. Can also be an inspiration for own implementation:

https://github.com/tensorfoo/milkha

Also AAT , but draws maps, so requires network:

https://github.com/bailuk/AAT/

pstorch, 2020-10-07:
A pure offline version of OSM Dashboard is on its way. Let's see how long it takes to get it published on F-Droid.

pstorch, 2020-10-14:
The pure offline version of OSM Dashboard is now available on F-Droid: https://f-droid.org/en/packages/de.storchp.opentracks.osmplugin.offline/

dennisguse, 2020-10-14:
@PhantomO2 is the offline OSMDashboard sufficient for you?

PhantomO2, 2020-10-17:
@dennisguse yep. Btw thanks for this awesome app!!!

dennisguse, 2020-10-18:
@PhantomO2 Don't thank me - it is a community project; we are in this all together (incl. users like you who contribute their ideas/requirements).
I just revived the project and did some refactoring ;)

closed 2020-10-18 by dennisguse