OpenTracksApp/OpenTracks #413: Crash after exporting train into map

petrhejna, 2020-09-05:
**Describe the bug**
Exporting track via "map" button in top left causes craah.

**To Reproduce**
1. Go to Track Detail
2. Click on Map Button
3. Doesnt matter which format 
4. Error appears (but sometimes, no error but nothing happens).

[image]
[image]

[image]

**Technical information**
 - Device: [One Plus 7 Pro]
 - OS: [e.g. LineageOS 16]
 - Version [v3.9.3]

dennisguse, 2020-09-05:
Hopefully fixed in https://github.com/OpenTracksApp/OpenTracks/issues/411 aka last night.
Released with v3.9.4 (already on PlayStore, but not yet on F-Droid).

Can you give it a try?

petrhejna, 2020-09-05:
Ah, good to hear that. Thanks. I use F-Droid only. So I'll try it once it there. 

:)

dennisguse, 2020-09-05:
Spoiler alert: I know - otherwise I would have gotten a notification from the PlayStore as they collect the stacktraces of all app crashes.
Privacy at it's best.

dennisguse, 2020-09-07:
Update just reached fdroid. @petrhejna can you check if this works for you?

petrhejna, 2020-09-07:
Works like a charm. Thank you:! :)

closed 2020-09-07 by petrhejna

dennisguse, 2020-09-08:
Thanks for testing and sorry for the inconvenience...