OpenTracksApp/OpenTracks #415: [Help] Help with an particular cadence sensor

ebellumat, 2020-09-05:
Hello Guys,

My name is Erasmo, I'm a developer, I need some help with a bluetooth device.

https://www.amazon.com/MEILAN-Wireless-Bluetooth-Computer-Waterproof/dp/B07QCPLVFC

I connect to him but I can't parse the retrieved data from this little device.

These device returns UInt8Array data.  

Anyone can help me to retrieve the cadence data from this device?

I want to get the RPM but i don't understand the returned values to convert.

Regards, Erasmo.

[image]
[image]

dennisguse, 2020-09-05:
I have no clue what kind of device this is, but the Bluetooth LE standard Cadence and Cycling sensor decoding of OpenTracks is here: https://github.com/OpenTracksApp/OpenTracks/blob/a1646e1b76f9bfabbc5622bca712c1a3b463f6c7/src/main/java/de/dennisguse/opentracks/util/BluetoothUtils.java#L85

You can also find the link to the standard.

Quick summary:
* timer (usually starts at 0 and is relative to device startup?)
* "wheel" rotation counter
=> Based upon this, you have to compute the cadence.
=> And Bluetooth specifies that these devices can transmit crank and wheel in the same message.

PS: This is not a question about OpenTracks, right?
PPS: Please label with "question" :)

dennisguse, 2020-09-05:
By the way the values that you send look okay.
This sensor sends "wheel".
And you can use this test in OpenTracks to manually decode the array: https://github.com/OpenTracksApp/OpenTracks/blob/f1281116501c4255159bba2eecd3908e1eaf5f21/src/androidTest/java/de/dennisguse/opentracks/util/BluetoothUtilsTest.java#L54

ebellumat, 2020-09-05:
> By the way the values that you send look okay.
> This sensor sends "wheel".
> And you can use this test in OpenTracks to manually decode the array: [...]

Thanks for your help! But, how did you know that this sends "wheel"? If you found any documentation about this device, please send to me, please!

dennisguse, 2020-09-05:
I dont have documentation for this device: it just looks like the Bluetooth LE standard ;)
A link to the spec is in the source code of OpenTracks.

PS: the first number is 2 and that means it is a wheel sensor ;)

ebellumat, 2020-09-05:
I think my code is finally work, thank you!

You help me a lot, i was measuring things using Javascript and now i will port this to a game in Unity (C#)

dennisguse, 2020-09-05:
You are welcome!

closed 2020-09-05 by dennisguse