OpenTracksApp/OpenTracks #418: Can we get reproducible builds?

dennisguse, 2020-09-07:
Follow-up to the discussion on Mastodon https://chaos.social/@dennisguse/104818490105850432

dennisguse, 2021-05-21:
Waiting for new builds on https://verification.f-droid.org

dennisguse, 2022-05-29:
A lot of details: https://f-droid.org/docs/Reproducible_Builds/

dennisguse, 2024-11-30:
OSMDashboard and also the Bahnhofsfoto App are now reproducible (thanks @pstorch)

Things that were done: 
1. https://github.com/RailwayStations/RSAndroidApp/commit/411426ebe54e32538aa0f0ac8285ef38a2a30dfb
2. https://gitlab.com/fdroid/fdroiddata/-/merge_requests/16486

For OpenTracks, we will likely publish reproducible builds under a new package name and keep both versions in the F-Droid store.
This allows current user to migrate whenever they want while new users can directly use the reproducible version.
Otherwise, we would force existing users to re-install incl. backup and restore - this has the tendency to be error prone...

dennisguse, 2024-12-03:
Could it be that the builds are already reproducible?
At least https://verification.f-droid.org reports that everything since versionCode 5683 could be verified.
Neat :D

For example: https://verification.f-droid.org/de.dennisguse.opentracks_5683.apk.json

dennisguse, 2024-12-25:
PR for F-Droid created: https://gitlab.com/fdroid/fdroiddata/-/merge_requests/17921

However, F-Droid seems to have trouble with checkupdates by identifying the new package name.
In OpenTracks this is implemented using buildTypes (with applicationIdSuffix).
OSMDashboard uses product flavors - and this works.

dennisguse, 2024-12-27:
@pstorch I saw that multiple fastlane translations were used for OSMDashboard.
Do you know where to find the documentation for this?

I would love to change the short summary for the irreproducible flavor.

pstorch, 2024-12-27:
@dennisguse no real documentation, but I got this ticket back then to fix the fastlane structure:
https://github.com/OpenTracksApp/OSMDashboard/issues/436

dennisguse, 2024-12-28:
PR to F-Droid was merged :partying_face:

dennisguse, 2025-01-03:
Anddddd it's live: https://f-droid.org/packages/de.dennisguse.opentracks.playstore

dennisguse, 2025-04-24:
@leoheitmannruiz thanks!
At some point this year, I will have time for this...

dennisguse, 2025-06-19:
i decided against separate fastlane structures as this will introduce differences in translations.
Not really that big of a benefit for explaining reproducible builds.

closed 2025-06-19 by dennisguse