OpenTracksApp/OpenTracks #420: F-Droid has trouble installing in Android 11 with Target SDK

dennisguse, 2020-09-09:
More information is available here: https://gitlab.com/fdroid/fdroidserver/-/issues/827

We have the option to downgrade to 29 as we do not use any 30 feature except that Android enforces more limitations like querying for other installed apps.
From a privacy perspective this is nice, but not that important for OpenTracks.

rgmf, 2020-09-09:
Downgrade to 29 is a good solution for now. When we add 30 features we'll see.

dennisguse, 2020-09-09:
Seemingly people have trouble #421.

duggalsu, 2020-09-09:
Downgrade is a good solution if it works on fdroid for now.

dennisguse, 2020-09-15:
We downgraded for v3.10.0 and wait until Fdroid fixes the issue.

ctennenhouse, 2020-09-16:
What is version 29? I only see 3.9.5, 3.9.4, 3.9.3 on F-droid.

dennisguse, 2020-09-16:
That is the targetSDKversion that OpenTracks is compiled for incl 3.9.5 it was 30 and due to the Fdroid issue 3.10.0 is 29.
But this version is not available via Fdroid - probably tomorrow. You can use the PlayStore version until then.

aaronaxvig, 2020-09-22:
3.10 is still not available on F-Droid.

dennisguse, 2020-09-22:
I know, but OpenTracks was built yesterday. So we are waiting for the signing. Can't do anything about this.

aaronaxvig, 2020-09-22:
Oh well, I ended up buying it on the Play Store so I guess you get a coffee out of it!

dennisguse, 2020-09-22:
@aaronaxvig FYI: the package name is different, so you can install both versions in parallel.

pstorch, 2020-09-23:
The fdroid buildserver was fixed to properly sign the APK with targetsdk 30. The OSMDashboard is in the current buildcycle with targetsdk30. If this is published, maybe someone can test it with Android 11. If that works we can upgrade the main application to targetsdk30 again also.

aaronaxvig, 2020-09-23:
I would be happy to test out new builds of either program if someone can point out when the new builds are available.  It is unfortunate that the builds take days/weeks to churn.

pstorch, 2020-09-24:
OSMDashboard is available on F-Droid in Version 2.4.1 (15) which is targeting sdk30.
@aaronaxvig would be nice if you could give it a try.

aaronaxvig, 2020-09-24:
Version 2.4.1 of OSMDashboard installs and works via F-Droid.

Also version 3.10 was available in F-Droid and installs and runs fine.

pstorch, 2020-09-24:
@aaronaxvig thanks for the feedback. Have you had to uninstall the previous version of OSMDashboard?

aaronaxvig, 2020-09-24:
> 
> 
> @aaronaxvig thanks for the feedback. Have you had to uninstall the previous version of OSMDashboard?

I'm pretty sure that I didn't uninstall it; in fact I was not able to find any way to interact with my previously installed versions of OpenTracker or OSMDashboard after my phone updated to Android 11.  They did not appear in the app list nor show as installed in F-Droid.

Once I installed the new version of each app via F-Droid, they had their old settings/data still in place (OpenTracker did not show the data from the Play Store version of OpenTracker, which I understand is expected because they are different namespaces).  For instance OpenTracker had some tracks from before the Android 11 update, and OSMDashboard has some configurations of offline maps still in there.

dennisguse, 2020-09-24:
Cool - so we survived this nasty thing then!
@aaronaxvig Thanks for testing!

closed 2020-09-24 by dennisguse