OpenTracksApp/OpenTracks #421: Cannot install on Android 11

duggalsu, 2020-09-09:
The app cannot be installed on Android 11 via Google Play Store OR F-droid. Installing from apk gives the following adb log - 

adb: failed to install ../de.dennisguse.opentracks_3584.apk: Failure [INSTALL_PARSE_FAILED_NO_CERTIFICATES: Scanning Failed.: No signature found in package of version 2 or newer for package de.dennisguse.opentracks]

This may be helpful - https://stackoverflow.com/questions/2914105/what-is-install-parse-failed-no-certificates-error

duggalsu, 2020-09-09:
Found out from another issue that it is a problem with fdroid - https://gitlab.com/fdroid/fdroidserver/-/issues/827

closed 2020-09-09 by duggalsu

dennisguse, 2020-09-09:
Weird the PlayStore version should also be signed with v1 and v2. I will check after work.

duggalsu, 2020-09-09:
I haven't tested  the play store version

dennisguse, 2020-09-09:
@duggalsu Could you try downgrading the targetSdkVersion in build.gradle to 29 and check if this can installed on your phone?
I don't have device to test and it would be great to know if this prevents the issue for now.

duggalsu, 2020-09-09:
Will try do it tomorrow.  Shall I push the changes if it works?